adafruit/Adafruit_nRF52_Arduino #183: MIDI central device notification problem

jardous, 2018-09-07:
Hello,

I have two Feathers nRF52. One is a peripheral programmed with the examples/blemidi and it works fine when I connect from Linux or iPad.

Unfortunately it seems that setting notification does not seem to work. The BLEClientCharacteristic.enableNotify() does not seem to have any effect and the actual data notification are not sent at all.

Here is sketch I use:
```c++
#include <bluefruit.h>

BLEClientService        midiService(BLEMIDI_UUID_SERVICE);
BLEClientCharacteristic midiChar(BLEMIDI_UUID_CHR_IO);
 
void setup()
{
  Serial.begin(115200);
 
  Serial.println("Bluefruit52 Central Custom HRM Example");
  Serial.println("--------------------------------------\n");
 
  // Initialize Bluefruit with maximum connections as Peripheral = 0, Central = 1
  // SRAM usage required by SoftDevice will increase dramatically with number of connections
  Bluefruit.begin(0, 1);
 
  Bluefruit.setName("Bluefruit52 Central");
 
  // Initialize HRM client
  midiService.begin();
 
  // Initialize client characteristics of HRM.
  // Note: Client Char will be added to the last service that is begin()ed.
  //bslc.begin();
 
  // set up callback for receiving measurement
  midiChar.setNotifyCallback(midi_notify_callback);
  midiChar.begin();
 
  // Increase Blink rate to different from PrPh advertising mode
  Bluefruit.setConnLedInterval(250);
 
  // Callbacks for Central
  Bluefruit.Central.setDisconnectCallback(disconnect_callback);
  Bluefruit.Central.setConnectCallback(connect_callback);
 
  /* Start Central Scanning
   * - Enable auto scan if disconnected
   * - Interval = 100 ms, window = 80 ms
   * - Don't use active scan
   * - Filter only accept HRM service
   * - Start(timeout) with timeout = 0 will scan forever (until connected)
   */
  Bluefruit.Scanner.setRxCallback(scan_callback);
  Bluefruit.Scanner.restartOnDisconnect(true);
  Bluefruit.Scanner.setInterval(160, 80); // in unit of 0.625 ms
  Bluefruit.Scanner.filterUuid(midiService.uuid);
  Bluefruit.Scanner.useActiveScan(false);
  Bluefruit.Scanner.start(0);                   // // 0 = Don't stop scanning after n seconds
}
 
void loop()
{
  // do nothing
}
 
void scan_callback(ble_gap_evt_adv_report_t* report)
{
  // Connect to device with HRM service in advertising
  Bluefruit.Central.connect(report);
}
 
void connect_callback(uint16_t conn_handle)
{
  Serial.println("Connected");
  Serial.print("Discovering HRM Service ... ");
 
  // If HRM is not found, disconnect and return
  if ( !midiService.discover(conn_handle) )
  {
    Serial.println("Found NONE");
 
    // disconect since we couldn't find HRM service
    Bluefruit.Central.disconnect(conn_handle);
 
    return;
  }
 
  // Once HRM service is found, we continue to discover its characteristic
  Serial.println("Found it");
 
  
  Serial.print("Discovering Measurement characteristic ... ");
  if ( !midiChar.discover() )
  {
    // Measurement chr is mandatory, if it is not found (valid), then disconnect
    Serial.println("not found !!!");  
    Serial.println("Measurement characteristic is mandatory but not found");
    Bluefruit.Central.disconnect(conn_handle);
    return;
  }
  Serial.println("Found it");
 
  // Reaching here means we are ready to go, let's enable notification on measurement chr
  if ( midiChar.enableNotify() )
  {
    Serial.println("Ready to receive MIDI");
  }else
  {
    Serial.println("Couldn't enable notify for MIDI. Increase DEBUG LEVEL for troubleshooting");
  }
}

void disconnect_callback(uint16_t conn_handle, uint8_t reason)
{
  (void) conn_handle;
  (void) reason;
 
  Serial.println("Disconnected");
}
 
void midi_notify_callback(BLEClientCharacteristic* chr, uint8_t* data, uint16_t len)
{
    Serial.print("MIDI: ");

    char value[64];
    memcpy(value, data, len);
    value[len] = 0;
 
    Serial.println(value);
}
```

jardous, 2018-09-10:
update: it looks like there is a problem writing to the client characteristic in the writeCCCD(). The Server (peripheral) does not call the write event handler at all (BLECharacteristic::_eventHandler)

jardous, 2018-09-11:
BLECentral seems to not send BLE_GAP_EVT_CONN_PARAM_UPDATE. Difference in logs on the Peripheral:

Connecting from iPad:

[BLE   ] BLE_GAP_EVT_CONNECTED : Conn Handle = 0
[GAP   ] Conn Interval= 30.000000
[BLE   ] BLE_GAP_EVT_CONN_PARAM_UPDATE : Conn Handle = 0
[GAP   ] Conn Interval= 15.000000
[BLE   ] BLE_GAP_EVT_SEC_INFO_REQUEST : Conn Handle = 0
[BOND  ] Load Keys from file /adafruit/bond/prph/35db
[BLE   ] BLE_GAP_EVT_CONN_SEC_UPDATE : Conn Handle = 0
[BOND  ] Load CCCD from file /adafruit/bond/prph/35db
[CHR   ] Notify 5 bytes
[CHR   ] Notify 5 bytes
[BLE   ] BLE_GATTS_EVT_HVN_TX_COMPLETE : Conn Handle = 0
[BLE   ] BLE_GATTS_EVT_HVN_TX_COMPLETE : Conn Handle = 0
[BLE   ] BLE_GATTS_EVT_EXCHANGE_MTU_REQUEST : Conn Handle = 0
[GAP   ] ATT MTU is changed to 185
[CHR   ] Notify 5 bytes
[CHR   ] Notify 5 bytes

... notifications kept sending ...

from the attached example:
[BLE   ] BLE_GAP_EVT_CONNECTED : Conn Handle = 0
[GAP   ] Conn Interval= 30.000000

the code hangs and it never gets the BLE_GAP_EVT_CONN_PARAM_UPDATE event

hathach, 2018-09-11:
Thanks for posting your progress of troubleshooting. Unfortunately, I am too busy with nrf52840 release to help you out for now.

jardous, 2018-09-11:
you are welcome, but I am afraid, that I have reached my limits here. Please let me know, as soon as you have time to look into this...

jps2000, 2018-09-18:
It is probably a similar problem  in the central custom HRM example.
Working with a polar H7 peripheral there comes abot after 30 sec
[BLE ] BLE_GAP_EVT_CONN_PARAM_UPDATE_REQUEST : Conn Handle = 0
which is not handled correctly. This results in a disconnect evoked by the peripheral
[BLE ] BLE_GAP_EVT_DISCONNECTED : Conn Handle = 0
Although it re- connects immediately after, the  uninterrupted transmission  of RR intervals  is not possible with that behavior.
see also https://forums.adafruit.com/viewtopic.php?f=53&t=140841

jardous, 2018-09-18:
@jps2000 yes, Peter. It seems to be the problem even with the example UART central device... Works well if I use my iPad as a central device so I suspect it is really a bug in the central lib.

I hope this will be solved soon because it is blocking me and I already spent too much time trying to fix it without success.

jps2000, 2018-09-18:
Did not say that, neither meant that.  Too many platforms for too little people. Its always a trade-off between width and depth in a business. Thats likely not your decision. Own problems are always considered major.....

jardous, 2018-09-27:
@hathach do you have any idea when approximately you finish the Circuitpython and Arduino running on nrf52840, and you can find some time to look into this? Thank you.

hathach, 2018-09-27:
about a few more weeks if things go well I guess.

jps2000, 2018-11-16:
@hathach  Hope things go well .Have seen in the forum hat you are back now. Any chance to get this issue solved in due time?

hathach, 2018-11-16:
@jps2000 yeah, indeed, 0.9.0 release has give me a bit more time. I am currently working on usb stack to better support circuitpython. It won't take much time, afterwards, I could then focus on troubleshooting the rest of this repo issue. Thank you for your patient.

hathach, 2019-01-04:
@jardous this issue is much easier than I thought. MIDI device require pairing to work (check Apple BLE Midi specs). Without pairing, the write simply failed. SD should return with error but look like it doesn't. Add this line of code to your central, It will be able to enable cccd and receive data from prph

```
  Serial.println("Found it");

  // Request to pair
  if ( !Bluefruit.Gap.requestPairing(conn_handle) )
  {
    Serial.print("Failed to paired");
    return;
  }
  Serial.println("Paired");

  // Reaching here means we are ready to go, let's enable notification chr
  if ( midiChar.enableNotify() )
 ``

jardous, 2019-01-04:
@hathach well - it really seems to work! Thanks

hathach, 2019-01-04:
@jardous  When you are positive, it is resolved, please close the issue.

jardous, 2019-01-04:
@hathach Should the examples be updated first?

hathach, 2019-01-04:
you mean the blemidi prph example, it is good enough as it is. Central should be the one to request pairing after discovering.

hajdbo, 2019-01-08:
@hathach @jardous I received two Feathers Express 52840 today and tried:

- : one with blemidi example, one with the blemidicentral described in the first comment (without the Bluefruit.Gap.requestPairing(conn_handle) line described in https://github.com/adafruit/Adafruit_nRF52_Arduino/issues/183#issuecomment-451329897 : Failed (as expected, I guess)
- one with blemidi example, one with the blemidicentral patched with the Bluefruit.Gap.requestPairing(conn_handle) line described in https://github.com/adafruit/Adafruit_nRF52_Arduino/issues/183#issuecomment-451329897 : Success (as expected)

*But* when I use a Akai LPK 25 wireless keyboard (BLE), Bluefruit.Gap.requestPairing() never returns.

The original sketch without Bluefruit.Gap.requestPairing() works fine and receives all MIDI traffic from the Akai keyboard.

jardous, 2019-01-08:
I have updated the central mode based on changes from @teafella 
check it out here: https://github.com/jardous/Adafruit_nRF52_Arduino

hathach, 2019-01-09:
@hajdbo thanks for your feedback, probably there is issue with pairing with your wireless keyboard, can you enable debug level = 2 then post the debug log here.

> I have updated the central mode based on changes from @teafella
> check it out here: https://github.com/jardous/Adafruit_nRF52_Arduino

Thanks, that will help other testing out the central midi quickly (including me)

hajdbo, 2019-01-09:
@hathach here is the debug=2 log for the Akai LPK 25 wireless midi keyboard (BLE) connecting to the Central Feathers Express 52840 with the  Bluefruit.Gap.requestPairing()  line.
It uses Arduino 1.8.8 and the current released version of the Adafruit nrf52 board support.

[nrf52840centralmidible-vs-akai-lpk25.txt](https://github.com/adafruit/Adafruit_nRF52_Arduino/files/2741266/nrf52840centralmidible-vs-akai-lpk25.txt)

The relevant section is here. No logs after this, even after pushing keyboard keys.
```
Connected
Discovering MIDI Service ... [DISC  ] [SVC] Handle start = 1
[BLE   ] BLE_GATTC_EVT_PRIM_SRVC_DISC_RSP : Conn Handle = 0
[DISC  ] [SVC] Service Count: 1
[DISC  ] [SVC] Found 0x0E5A, Handle start = 28, end = 32
-----------------
Found it
Discovering Measurement characteristic ... [DISC  ] [CHR] Handle start = 28, end = 32
[BLE   ] BLE_GATTC_EVT_CHAR_DISC_RSP : Conn Handle = 0
[DISC  ] [CHR] Characteristic Count: 1
[DISC  ] [CHR] Found 0xE5DB, handle = 30
-----------------
[DISC  ] [DESC] Handle start = 31, end = 32
[BLE   ] BLE_GATTC_EVT_DESC_DISC_RSP : Conn Handle = 0
[DISC  ] [DESC] Descriptor Count: 2
[DISC  ] [DESC] Descriptor 0: uuid = 0x2902, handle = 31
[DISC  ] [DESC] Descriptor 1: uuid = 0x2901, handle = 32
[DISC  ] Found CCDD: handle = 31
Found it
[BLE   ] BLE_GAP_EVT_AUTH_STATUS : Conn Handle = 0
void BLEGap::_eventHandler(ble_evt_t*): 348: status->auth_status = 0x85
[BLE   ] BLE_GAP_EVT_CONN_PARAM_UPDATE_REQUEST : Conn Handle = 0
```

hajdbo, 2019-01-09:
Compare it with the debug log without the Bluefruit.Gap.requestPairing(conn_handle) call:

```
Bluefruit52 Central Custom MIDI Example
--------------------------------------

[CFG   ] SoftDevice's RAM requires: 0x20002478

BSP Library : 0.9.2
Bootloader  : s140 6.1.1 r1
Serial No   : 5FD32B748BD7E4E1

--------- SoftDevice Config ---------
Max UUID128     : 10
ATTR Table Size : 2048
Service Changed : 1
Central Connect Setting
  - Max MTU         : 23
  - Event Length    : 3
  - HVN Queue Size  : 1
  - WrCmd Queue Size: 1

--------- BLE Settings ---------
Name            : Bluefruit52 Central
Max Connections : Peripheral = 0, Central = 4 
Address         : E2:A7:3A:7C:77:A8 (Static)
TX Power        : 0 dBm
Conn Intervals  : min = 20.00 ms, max = 30.00 ms
Conn Timeout    : 2000.00 ms
Central Paired Devices: 
  8bec : Bluefruit52 MIDI (99 bytes)

[BLE   ] BLE_GAP_EVT_ADV_REPORT : Conn Handle = 65535
[BLE   ] BLE_GAP_EVT_ADV_REPORT : Conn Handle = 65535
[BLE   ] BLE_GAP_EVT_ADV_REPORT : Conn Handle = 65535
[BLE   ] BLE_GAP_EVT_ADV_REPORT : Conn Handle = 65535
[BLE   ] BLE_GAP_EVT_ADV_REPORT : Conn Handle = 65535
[BLE   ] BLE_GAP_EVT_ADV_REPORT : Conn Handle = 65535
[BLE   ] BLE_GAP_EVT_ADV_REPORT : Conn Handle = 65535
[BLE   ] BLE_GAP_EVT_ADV_REPORT : Conn Handle = 65535
[BLE   ] BLE_GAP_EVT_ADV_REPORT : Conn Handle = 65535
[BLE   ] BLE_GAP_EVT_ADV_REPORT : Conn Handle = 65535
[BLE   ] BLE_GAP_EVT_ADV_REPORT : Conn Handle = 65535
[BLE   ] BLE_GAP_EVT_CONNECTED : Conn Handle = 0
Connected
Discovering MIDI Service ... [DISC  ] [SVC] Handle start = 1
[BLE   ] BLE_GATTC_EVT_PRIM_SRVC_DISC_RSP : Conn Handle = 0
[DISC  ] [SVC] Service Count: 1
[DISC  ] [SVC] Found 0x0E5A, Handle start = 28, end = 32
-----------------
Found it
Discovering Measurement characteristic ... [DISC  ] [CHR] Handle start = 28, end = 32
[BLE   ] BLE_GATTC_EVT_CHAR_DISC_RSP : Conn Handle = 0
[DISC  ] [CHR] Characteristic Count: 1
[DISC  ] [CHR] Found 0xE5DB, handle = 30
-----------------
[DISC  ] [DESC] Handle start = 31, end = 32
[BLE   ] BLE_GATTC_EVT_DESC_DISC_RSP : Conn Handle = 0
[DISC  ] [DESC] Descriptor Count: 2
[DISC  ] [DESC] Descriptor 0: uuid = 0x2902, handle = 31
[DISC  ] [DESC] Descriptor 1: uuid = 0x2901, handle = 32
[DISC  ] Found CCDD: handle = 31
Found it
Ready to receive MIDI
[BLE   ] BLE_GATTC_EVT_WRITE_CMD_TX_COMPLETE : Conn Handle = 0
[BLE   ] BLE_GAP_EVT_CONN_PARAM_UPDATE_REQUEST : Conn Handle = 0
[BLE   ] BLE_GATTC_EVT_HVX : Conn Handle = 0
MIDI: A7DC903C42
[BLE   ] BLE_GATTC_EVT_HVX : Conn Handle = 0
MIDI: A8D7803C0
[BLE   ] BLE_GATTC_EVT_HVX : Conn Handle = 0
MIDI: AEA4903C44
[BLE   ] BLE_GATTC_EVT_HVX : Conn Handle = 0
MIDI: AEF8803C0
[BLE   ] BLE_GATTC_EVT_HVX : Conn Handle = 0
MIDI: B1BF903E37
[BLE   ] BLE_GATTC_EVT_HVX : Conn Handle = 0
MIDI: B1F5803E0
[BLE   ] BLE_GATTC_EVT_HVX : Conn Handle = 0
MIDI: BA90903B40
[BLE   ] BLE_GATTC_EVT_HVX : Conn Handle = 0
MIDI: BAFD803B0
```

hathach, 2019-01-09:
@hajdbo the log indicate the pairing error, the code 0x85 means your keyboard does not support pairing. I just checked the MIDI specs again: they removed the pairing requirement, and only **recommend** instead of **must** support pairing now. 

`
#define BLE_GAP_SEC_STATUS_PAIRING_NOT_SUPP       0x85  /**< Pairing not supported.  */
`
```
[BLE   ] BLE_GAP_EVT_AUTH_STATUS : Conn Handle = 0
void BLEGap::_eventHandler(ble_evt_t*): 348: status->auth_status = 0x85
```

The **RIGHT** solution is of course you have to check the midi characteristics read/write encryption level to see if pairing is needed or not. Such as query function is not implemented yet. I just open an new issue for that #213 . Meanwhile you can just simply ignore the failed pairing

```
  // Request to pair
  if ( !Bluefruit.Gap.requestPairing(conn_handle) )
  {
    Serial.print("Failed to paired");
    // return; // comment out return to ignore failed pairing request.
  }
  Serial.println("Paired");
```