adafruit/Adafruit_nRF52_Arduino #184: Cannot Verify Code - No Such File

dwilday, 2018-09-19:
Following the install directions for the mac, when I attempt to verify any code (even a simple blink) I receive the following error.

```
fork/exec /Users/myusername/Library/Arduino15/packages/adafruit/tools/gcc-arm-none-eabi/5_2-2015q4/bin/arm-none-eabi-g++: no such file or directory
Error compiling for board Adafruit Bluefruit nRF52 Feather.
```

Likewise, if I attempt to burn the bootloader I received the following:

```
[24539] Error loading Python lib '/var/folders/rw/rvc7zzs53n7cjnc8n8ckfsw93l3dhz/T/_MEI8YOy6A/Python': dlopen: dlopen(/var/folders/rw/rvc7zzs53n7cjnc8n8ckfsw93l3dhz/T/_MEI8YOy6A/Python, 10): Symbol not found: _futimens
  Referenced from: /var/folders/rw/rvc7zzs53n7cjnc8n8ckfsw93l3dhz/T/_MEI8YOy6A/Python (which was built for Mac OS X 10.13)
  Expected in: /usr/lib/libSystem.B.dylib
 in /var/folders/rw/rvc7zzs53n7cjnc8n8ckfsw93l3dhz/T/_MEI8YOy6A/Python
Error while burning bootloader.
```

Any idea what's going on?

hathach, 2018-09-24:
are you using the older version of macos. The pre-builtin binary `adafruit-nrfutil` is produced in the macos 10.13, something maybe missing if you use older macos. Can you follow  my suggestion here to product the `adafruit-nrfutil` on your machine, and replace the one in bsp

https://forums.adafruit.com/viewtopic.php?f=57&t=140903&p=696619&hilit=nrf52#p696619

hathach, 2018-11-19:
close due to lack of user info

closed 2018-11-19 by hathach

TechplexEngineer, 2019-05-18:
I am on osx 10.12.6 and I had to follow the steps in the linked post to get this to work.