adafruit/Adafruit_nRF52_Arduino #185: Regular disconnects every 70 seconds when connecting Feather NRF52 to speed sensor

HiFiJ, 2018-09-22:
Hi everybody,

this is just the reformat of my forum post to create a github issue, as I get the impression that the problem I am seeing might indeed be a bug/missing implementation in BlueFruit. IMHO if this is the case, this would restrict the usability of the nRF52 Feather pretty severely (as I believe applications like the one I am planning might be pretty typical). Thank you very much for/if taking a look into it!
So here goes:

Hi everybody,

thanks for helping each other and thanks to Adafruit for this cool little board (Feather NRF52).
I am currently building a bike computer head unit consisting of the feather, a tiny 300mAh LiPo and a small ePaper display (which works well already); later on all fitted in a 3d printed case.
The sensor on the bike in use is commercially available, it is a Wahoo Speed Sensor implementing the Bluetooth CSC specifications.

[https://eu.wahoofitness.com/devices/bike-sensors/bluetooth-speed-sensor?stay]
[https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.service.cycling_speed_and_cadence.xml]

I successfully implemented a working prototype using the Espruino Pixl.js (my working code example can be found here [http://forum.espruino.com/conversations/323575/]), managing to connect to the sensor without unwanted disconnects.

Now, switching to the Feather for size reasons and easier interfacing of the ePaper display, I have the following code (copied directly from my IDE, this exact code exhibits my problem), adapted from the HRM API example:

```
#include <bluefruit.h>

BLEClientService speedsensorservice(UUID16_SVC_CYCLING_SPEED_AND_CADENCE);
BLEClientCharacteristic speedsensorcharacteristic(UUID16_CHR_CSC_MEASUREMENT);
uint32_t millis_connect,millis_disconnect; float connect_duration;

void setup()
{
  Serial.begin(115200);

  Bluefruit.begin(0, 1); //peripheral, central
  Bluefruit.setName("Bluefruit52 Central");

   speedsensorservice.begin();
  speedsensorcharacteristic.setNotifyCallback(csc_notify_callback);
  speedsensorcharacteristic.begin();

  Bluefruit.Central.setDisconnectCallback(disconnect_callback);
  Bluefruit.Central.setConnectCallback(connect_callback);

  Bluefruit.Scanner.setRxCallback(scan_callback);
  Bluefruit.Scanner.restartOnDisconnect(true);
  Bluefruit.Scanner.setInterval(160, 80); // in unit of 0.625 ms
  Bluefruit.Scanner.filterUuid(speedsensorservice.uuid);
  Bluefruit.Scanner.useActiveScan(false);
  Bluefruit.Scanner.start(60);                   // // 0 = Don't stop scanning after n seconds
}

void loop()
{
}

void scan_callback(ble_gap_evt_adv_report_t* report)
{
  Bluefruit.Central.connect(report);
}

void connect_callback(uint16_t conn_handle)
{
  millis_connect = millis();
  Serial.printf("Connected at %d\n", millis_connect);
  Serial.print("Discovering CSC Service ... ");

  if ( !speedsensorservice.discover(conn_handle) )
  {
    Serial.println("Found NONE");
    Bluefruit.Central.disconnect(conn_handle);
    return;
  }
  Serial.println("Found it");Serial.print("Discovering Measurement characteristic ... ");
  if ( !speedsensorcharacteristic.discover() )
  {
    // Measurement chr is mandatory, if it is not found (valid), then disconnect
    Serial.println("not found !!!");Serial.println("Measurement characteristic is mandatory but not found");
    Bluefruit.Central.disconnect(conn_handle);
    return;
  }
  Serial.println("Found it");

  if ( speedsensorcharacteristic.enableNotify() )
  {
    Serial.println("Ready to receive CSC Measurement value");
  }else
  {
    Serial.println("Couldn't enable notify for CSC Measurement. Increase DEBUG LEVEL for troubleshooting");
  }

  Bluefruit.printInfo();

}

void disconnect_callback(uint16_t conn_handle, uint8_t reason)
{
  (void) conn_handle;
  (void) reason;

  millis_disconnect = millis();
  Serial.printf("Disconnected due to reason nr. 0x%x\n", reason);
  Serial.printf("Disconnected at %d\n", millis_disconnect); connect_duration = (millis_disconnect - millis_connect)/1000;
  Serial.printf("Duration: %f sec\n", connect_duration);
}

void csc_notify_callback(BLEClientCharacteristic* chr, uint8_t* data, uint16_t len)
{
  uint16_t current_wheel_revolutions;
  if ( data[0] & bit(0) )
  {
    memcpy(&current_wheel_revolutions, data+1, 4);
  }

  Serial.printf("Wheel revolutions: %4d\n" ,current_wheel_revolutions);
}

```

This establishes a connection successfully and receives the CSC notifications via csc_notify_callback, but in contrast to the Espruino impementation suffers from recurrent disconnections every 70 seconds. This is the console output:

> [Starting] Opening the serial port - /dev/ttyUSB0
> [Info] Opened the serial port - /dev/ttyUSB0
> Connected at 1941
> Discovering CSC Service ... Found it
> Discovering Measurement characteristic ... Found it
> Ready to receive CSC Measurement value
> --------- SoftDevice Config ---------
> Max UUID128 : 0
> ATTR Table Size : 2048
> Service Changed : 0
> 
> --------- BLE Settings ---------
> Name : Bluefruit52 Central
> Max Connections : Peripheral = 0, Central = 4 
> Address : CF:86:12:67:1E:0B (Static)
> TX Power : 0 dBm
> Conn Intervals : min = 20.00 ms, max = 30.00 ms
> Peripheral Paired Devices: 
> 
> Central Paired Devices: 
> 
> 
> Wheel revolutions: 83
> ... (going on 70secs; receiving correct data as I did not revolve the sensor - revoloving the sensor does not change the disconnect interval)
> Wheel revolutions: 83
> 
> Disconnected due to reason nr. 0x13
> Disconnected at 71993
> Duration: 70.000000 sec
> Connected at 72327
> Discovering CSC Service ... Found it
> Discovering Measurement characteristic ... Found it
> Ready to receive CSC Measurement value
> --------- SoftDevice Config ---------
> Max UUID128 : 0
> ATTR Table Size : 2048
> Service Changed : 0
> 
> --------- BLE Settings ---------
> Name : Bluefruit52 Central
> Max Connections : Peripheral = 0, Central = 4 
> Address : CF:86:12:67:1E:0B (Static)
> TX Power : 0 dBm
> Conn Intervals : min = 20.00 ms, max = 30.00 ms
> Peripheral Paired Devices: 
> 
> Central Paired Devices: 
> 
> Wheel revolutions: 83
> Wheel revolutions: 83

Now, the disconnect reason is reported as 0x13 (GATT CONN TERMINATE PEER USER). The completely regular interval of 70 seconds reminds me of some kind of timeout triggering the disconnect (btw., not prolonged if using/revolving the sensor). I found very sparse hints as to the cause and only stumbled over these threads:

[(https://github.com/adafruit/Adafruit_nRF52_Arduino/issues/100)]
[https://devzone.nordicsemi.com/f/nordic-q-a/27066/ble-central-disconnect-after-exactly-65-5-sec (coincidentally started by Gordon Williams, the creator of Espruino - he seems to have solved this problem...)]

Can anyone shed some light on the reason for these disconnects happening and help me get a stable connection to the sensor? 
Thanks everyone for reading this long post and for any comments,

Joost

Here is the link to the forum post:
[https://forums.adafruit.com/viewtopic.php?f=57&t=141283]
where jps200pointed to a possible related post/issue: [https://github.com/adafruit/Adafruit_nRF52_Arduino/issues/183]

jps2000, 2018-12-30:
@hathach 
Any chance to get this solved in this decade  ?

hathach, 2019-01-04:
disconnect with 0x13 is `BLE_HCI_REMOTE_USER_TERMINATED_CONNECTION` which means the prph actively initiate disconnect procedure. Maybe it expects something that central didn't do. Did prph requires pairing/bonding to work in long term or requires a specific timing (conn interval etc ..). Can you try to call requestPairing() to initiate the pairing to see if it work better.

Analyzing your debug log doesn't show any abnormal events, since I don't have your exact device, I can't reproduce the issue, and there is little I could do to troubleshoot this besides giving hint.

hathach, 2019-01-07:
update: there is an update for central that handle connection params request update. You may want to test it on develop branch

hathach, 2019-02-15:
closed due to lack of user feedback

closed 2019-02-15 by hathach