adafruit/Adafruit_nRF52_Arduino #188: Serial.print and floating point within SoftwareTimer Callback

JDowns1971, 2018-09-26:
Serial.print(float) or Serial.println(float) within a SoftwareTimer callback causes a hardware crash. 

Serial.print(float) or Serial.println(float) within setup()  or loop() is fine.

If you convert the float to a non FPU then serial.print works fine.

hathach, 2018-09-26:
Software timer callback run within a minimal stack of timer task, printf or more complicated func will likely cause stack overflow. You either try to increase timer stack size in freertos config file or create a separated task for your work,

closed 2018-09-26 by hathach

JDowns1971, 2018-09-26:
Thank you for the quick response hathach.

increasing the timer stack from 100 to 120 fixed the issue for me.

Regards,