adafruit/Adafruit_nRF52_Arduino #195: fork and port Encoder library to replace current SwRotaryEncoder

hathach, 2018-11-07:
https://github.com/PaulStoffregen/Encoder

henrygab, 2020-01-12:
Hi Hathach, noticed you were looking for a QDEC library.

May I recommend the following?
https://github.com/SimpleHacks/QDEC

It's MIT-licensed, simple, state-machine based (no debouncing required), and has a set of examples showing usage from simplest loop() polling to interrupt-based usage.

Or, are you looking for something that specifically mimics the PaulStoffregen Encoder library API, but uses the QDEC peripheral to offload the state change detection?

hathach, 2020-01-15:
superb!! Thank you @henrygab, this looks very promising. Anything that get the Encoder works is good enough for me. Ideally if we can have QDEC prph working in the background it would be useful since hopefully it will detect states better than gpio. I will give it a try whenever  I could.