adafruit/Adafruit_nRF52_Arduino #197: Change Advertisement Name "on the fly"

Splitframe, 2018-11-08:
Hi,

I want to change the Advertisement name "on the fly".

Currently I do this on the disconnect callback:

`void prph_disconnect_callback(uint16_t conn_handle, uint8_t reason)
{
  (void) conn_handle;
  (void) reason;
  
  Bluefruit.Advertising.stop();
  Bluefruit.Advertising.clearData();
  Bluefruit.setName(cname);
  Bluefruit.ScanResponse.addName();
  Bluefruit.Advertising.start();
  
}`

The device is not visible afterwards though.

hathach, 2018-11-08:
you could just stop the advertising at anytime (not just disconnect callback). Then change the name and yes, clearData() and re-create the adv/scan_resp packet. Since this is not often used by most user, and it is rather trivial, we don't feel the need to implement this.

hathach, 2018-11-09:
I will give it a try, meanwhile can you enable debug level to 1 (or 2) to see if there is any error message

Splitframe, 2018-11-13:
I looked around a little, but I don't know how I enable the debug level.

hathach, 2018-11-13:
it is in in IDE menu called "Debug Mode"

Splitframe, 2018-11-14:
Thanks.
Here it is: https://pastebin.com/P8DVQdEe

hathach, 2018-11-19:
hi @Splitframe , you didn't re-set the name right. Maybe we didn't make it clear. Advertising and ScanResponse is 2 separate packet.
- Advertising: 31 bytes that broadcast by nrf52 to central, however it is too short for all information
- ScanReponse: another 31 bytes that is a response to central when it perform active scan (thus the name scan response).

In your above code, the service is fitted to the advertising packet, and the name is fit to Scan response. If you want to change the name, you only need to clear the scan response e.g  
 - Bluefruit.ScanResponse.clearData() and add back the name Bluefruit.ScanResponse.addName();
 - or clear both the Advertising and ScanResponse and call setupAdv() again

```C
void changeAdvertisementName() {
  Bluefruit.Advertising.stop();
  Bluefruit.Advertising.clearData();
  Bluefruit.ScanResponse.clearData(); // add this
  Bluefruit.setName(cname);
  setupAdv();
  Bluefruit.Advertising.start();
}
```

Splitframe, 2018-11-21:
That works, thanks you.

Sorry for the inconvenience.

hathach, 2018-11-21:
Glad that work for you

closed 2018-11-21 by hathach

aqibnasim, 2023-01-25:
Hi @hathach ,

I am trying to do the same procedure to change the name and getting the same. I am getting error when debug mode is enabled:

`bool BLEAdvertising::_start(uint16_t, uint16_t): 364: verify failed, error = NRF_ERROR_INVALID_LENGTH`

These are my device details:

```
18:57:16.979 -> [CFG   ] SoftDevice's RAM requires: 0x20002D30
18:57:22.012 -> Advertising is started
18:57:22.012 -> 
18:57:22.012 -> BSP Library : 1.0.1
18:57:22.012 -> Bootloader  : s140 6.1.1
18:57:22.012 -> Serial No   : AB3586AB30FBC471
18:57:22.012 -> 
18:57:22.012 -> --------- SoftDevice Config ---------
18:57:22.012 -> Max UUID128     : 10
18:57:22.012 -> ATTR Table Size : 4096
18:57:22.012 -> Service Changed : 1
18:57:22.012 -> Peripheral Connect Setting
18:57:22.012 ->   - Max MTU         : 23
18:57:22.012 ->   - Event Length    : 3
18:57:22.012 ->   - HVN Queue Size  : 1
18:57:22.012 ->   - WrCmd Queue Size: 1
18:57:22.012 -> 
18:57:22.012 -> --------- BLE Settings ---------
18:57:22.012 -> Name            : Bye world
18:57:22.012 -> Max Connections : Peripheral = 1, Central = 0 
18:57:22.012 -> Address         : C6:CA:9F:1A:75:3D (Static)
18:57:22.012 -> TX Power        : 4 dBm
18:57:22.012 -> Conn Intervals  : min = 20.00 ms, max = 30.00 ms
18:57:22.012 -> Conn Timeout    : 2000 ms
18:57:22.012 -> Peripheral Paired Devices:
```

CSC-Sendance, 2023-07-24:
> My code snippets:
> 
> ```
> void startAdv(void)
> {
>   // Advertising packet
>   Bluefruit.Advertising.addFlags(BLE_GAP_ADV_FLAGS_LE_ONLY_GENERAL_DISC_MODE);
>   Bluefruit.Advertising.addTxPower();
>   Bluefruit.ScanResponse.addName();
>   Bluefruit.Advertising.start();
> }
> ```
> 
> ... and the flow of code:
> 
> ```
>   Serial.println("Stopping Advertising");
>   Bluefruit.Advertising.stop();
> 
>   delay(5000);
>   
>   Bluefruit.Advertising.clearData();
>   Bluefruit.ScanResponse.clearData();
>   Bluefruit.setName("Hello world");
>   
>   Serial.println("Resuming Advertising");
>   startAdv();
>   delay(20000);
> 
>   Serial.println("Stopping Advertising");
>   Bluefruit.Advertising.stop();
> 
>   delay(5000);
> 
>   Bluefruit.Advertising.clearData();
>   Bluefruit.ScanResponse.clearData();
>   Bluefruit.setName("Bye world");
> 
>   Serial.println("Resuming Advertising");
>   startAdv();
>   delay(20000);
> ```

Hi, I had the same issue and fixed it in this PR: https://github.com/adafruit/Adafruit_nRF52_Arduino/pull/784 .

@hathach would be lovely if you could check it out!

commit bf504795 (2023-07-26): Merge pull request #784 from CSC-Sendance/master

FIX #197 : Replace varclr with arrclr for BLEAdvertising _data fields
commit dc1b3d75 (2023-07-26): FIX #197 : Replace varclr with arrclr for BLEAdvertising _data fields

Otherwise the data field gets corrupted (?) causing error msg "bool BLEAdvertising::_start(uint16_t, uint16_t): 364: verify failed, error = NRF_ERROR_INVALID_LENGTH" as mentioned in this comment: https://github.com/adafruit/Adafruit_nRF52_Arduino/issues/197#issuecomment-1403674667

see also https://devzone.nordicsemi.com/f/nordic-q-a/52573/sd_ble_gap_adv_set_configure-returns-error-nrf_error_invalid_length