adafruit/Adafruit_nRF52_Arduino #210: nrf52 recive nrf24

spa-sam, 2019-01-03:
How to receive data on nrf52 from nrf24?
Is it possible to add an example to the library?

ladyada, 2019-01-03:
not at this time, that's a completely new protocol to support, we have no plans to support it at this time but if someone is interested in submitting a PR we'd love to check it out!

hathach, 2019-01-04:
for reference if you want to add ESB support.
- https://github.com/NordicPlayground/nrf51-ble-micro-esb-uart
- https://devzone.nordicsemi.com/b/blog/posts/running-micro-esb-concurrently-with-ble

SolidGeek, 2019-07-18:
Would love for this to be implemented (could possible do it myself), as I have a few uses for it with my projects. Has anyone given this some beginning thoughts, as I am not sure where to start. Any guidance?

hathach, 2020-02-03:
closed due to lack of activities

closed 2020-02-03 by hathach