adafruit/Adafruit_nRF52_Arduino #214: add USB MSC for feather nrf52840

hathach, 2019-01-10:


jeremypoulter, 2019-01-10:
FYI I did some initial work on getting this integrated into PluggableUSB on the SAMD21. Not sure if it is useful or not as I never completed it, ran out of time and knowledge.

hathach, 2019-01-10:
@jeremypoulter Thanks for suggestion. The usb code is already done and running with pca10056 (nrf52840dk) with fatfs file system on top of its external 8MB flash  https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/libraries/FileSystem/src/ExternalFS.h

The issue with nrf52840 is the qspi flash device `GD25Q16C`, my current qspi driver somehow doesn't work with it. Unfortunately, I don't have any Feather nRF52840 on my hands to test with (need to wait a couple of months or so). This is a prue reminder for me to test and add support for it later on.

ladyada, 2019-01-10:
adding pluggable USB support would let us have MIDI and HID - but we're not sure if its the way we're planning to go!

hathach, 2019-05-13:
implemented by https://github.com/adafruit/Adafruit_TinyUSB_Arduino

closed 2019-05-13 by hathach