adafruit/Adafruit_nRF52_Arduino #218: Any way to secure against OTA DFU?

lbdroid, 2019-01-16:
Looking through the literature on these as carefully as I can, I've come across reference to a mode referred to as "OTA DFU", which can apparently be triggered by Nordic's nRF Toolbox application, allowing the storage memory to be rewritten over an unsecured bluetooth connection.

I've also seen indication that there is absolutely no security on this functionality at all: https://github.com/adafruit/Adafruit_nRF52_Arduino/issues/162#issuecomment-400563474

What I need to be able to do is disable this mode altogether, or in the very least, be able to program it with an update key that I (and only I) am in control of. This "feature" otherwise presents a security hole that makes it completely unsuitable for any application that has even fringe security related implications.

hathach, 2019-01-17:
This is indeed mostly bootloader implementation 
https://github.com/adafruit/Adafruit_nRF52_Bootloader

Bootloader OTA currently doesn't require device must be previously bonded/paired. It can be enforced to prevent unauthorized update, please also post your issue there. Or better submit an PR for that feature.

lbdroid, 2019-01-18:
Actually, it doesn't look like it is exclusively an issue with the bootloader.

Correct me if I'm wrong, but the OTA DFU process seems to work like this;
1) Normal running condition advertises DFU OTA service.
2) Connected devices calls on DFU service to reboot into OTA DFU mode.
3) Once in DFU OTA mode, connected device sends payload.

Killing this line stops DFU OTA service from starting; https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/libraries/Bluefruit52Lib/src/bluefruit.cpp#L438

It also seems that without DFU OTA service, it would be necessary to obtain physical access to the device in order to cause it to reboot into that mode.

hathach, 2019-01-18:
1. No
2. Yes
3. Yes

without dfu service the only way to trigger OTA is having physical access the the board which is rather pointless for OTA anyway. We don't provide API to exclude DFU service, so you need to modify BSP locally for your need (remove that line).

ASerbinski, 2019-01-19:
Perfect. Thank you.

lbdroid, 2019-01-19:
Perfect, thank you.

closed 2019-01-19 by lbdroid

hathach, 2019-02-01:
we have added enableOTA(bool) to enable/disable OTA service in application, will update Learn tutorial for this new API

henrygab, 2019-03-11:
Great to hear that an explicit call is now needed to enable OTA DFU.  Is this to be called as ```Bluefruit.enableOTA(true)```?  Can you confirm the default is to **_disable_** OTA DFU?

Thank you!

hathach, 2019-03-11:
by default OTA is enabled to be consistent with previous behavior of Bluefruit. Note: the change is on `develop` branch and haven't merged yet to master.

hathach, 2019-04-04:
I changed my mind, I separate DFU OTA service from Bluefruit.begin(), sketch must explicitly add dfu service. To be consistent, dfu ota service should be added first.

henrygab, 2019-04-04:
Wonderful, thank you!  This will remove the risk of any demos being interfered with by an insecure, on-by-default, OTA firmware update capability.  I was imagining the equivalent of a TV-be-gone remote, which just broadcast "do nothing" firmware to all devices that advertised DFU... Your choice to be secure-by-default is greatly appreciated!

tusharvb19, 2020-09-23:
Any progress?
I would like to have secure DFU OTA.