adafruit/Adafruit_nRF52_Arduino #222: Connecting/Reconnecting a peripheral to a central can corrupt the LittleFS FS where pairing information is stored.

jpconstantineau, 2019-01-29:
- Please provide details that help us to reproduce your issue
That's a hard one... generally disconnecting and reconnecting can cause this issue but doesn't happen all the time.  It's not a critical problem when a central (nrf52832) reconnects to a peripheral (nrf52832) but is an issue when a computer reconnects to a "faulty" peripheral.  It appears that a nrf52 central will re-negociate the pairing information.  However, a computer will retain the pairing information and keep retrying (as seen by the toggling back and forth between paired and connected on windows 10).

I suspect the issue occurs on the original pairing and not the reconnection.

  - Your Board and BSP version
       BSP Library : 0.9.3
       Bootloader  : s132 6.1.1 r1
  - Your OS: 
       Windows 10
  - Steps to reproduce, or better your (simplified) sketch if possible
Can't yet identify the steps to reproduce or the root cause. (will keep working on that)

  - Serial debug log: all at Debug Level 2:
Problem 1 (on reconnection - but not the actual root cause):

```
lfs error:494: Corrupted dir pair at 3 2
virtual bool LittleFS::rmdir_r(const char*): 373: verify failed, error = LFS_ERR_CORRUPT

```

Problem 2 (after restarting):

```
Peripheral Paired Devices: 
lfs error:494: Corrupted dir pair at 124 123
BluefruitFS::File LittleFS::_open_dir(const char*): 281: verify failed, error = LFS_ERR_CORRUPT

```

** When windows tries to reconnect, all we see is this (and it keeps repeating):**

```
[BLE   ] BLE_GAP_EVT_CONNECTED : Conn Handle = 0
[GAP   ] Conn Interval= 11.250000
[BLE   ] BLE_GAP_EVT_SEC_INFO_REQUEST : Conn Handle = 0
lfs error:494: Corrupted dir pair at 124 123
virtual BluefruitFS::File LittleFS::open(const char*, uint8_t): 313: verify failed, error = LFS_ERR_CORRUPT
[BLE   ] BLE_GAP_EVT_DISCONNECTED : Conn Handle = 0
[GAP   ] Disconnect Reason 0x13
```

** Even after loading the "Clear Bonds" program. I was left with the same problem:**

```
Bluefruit52 Clear Bonds Example
-------------------------------

[CFG   ] SoftDevice's RAM requires: 0x20002530
lfs error:494: Corrupted dir pair at 3 2
lfs error:494: Corrupted dir pair at 3 2
virtual bool LittleFS::mkdir(const char*): 339: verify failed, error = LFS_ERR_CORRUPT
lfs error:494: Corrupted dir pair at 3 2
lfs error:494: Corrupted dir pair at 3 2
virtual bool LittleFS::mkdir(const char*): 339: verify failed, error = LFS_ERR_CORRUPT

----- Before -----

lfs error:494: Corrupted dir pair at 3 2
virtual BluefruitFS::File LittleFS::open(const char*, uint8_t): 313: verify failed, error = LFS_ERR_CORRUPT

lfs error:494: Corrupted dir pair at 3 2
virtual BluefruitFS::File LittleFS::open(const char*, uint8_t): 313: verify failed, error = LFS_ERR_CORRUPT

lfs error:494: Corrupted dir pair at 3 2
virtual bool LittleFS::rmdir_r(const char*): 373: verify failed, error = LFS_ERR_CORRUPT
lfs error:494: Corrupted dir pair at 3 2
virtual bool LittleFS::mkdir(const char*): 339: verify failed, error = LFS_ERR_CORRUPT
lfs error:494: Corrupted dir pair at 3 2
virtual bool LittleFS::rmdir_r(const char*): 373: verify failed, error = LFS_ERR_CORRUPT
lfs error:494: Corrupted dir pair at 3 2
virtual bool LittleFS::mkdir(const char*): 339: verify failed, error = LFS_ERR_CORRUPT

----- After  -----

lfs error:494: Corrupted dir pair at 3 2
virtual BluefruitFS::File LittleFS::open(const char*, uint8_t): 313: verify failed, error = LFS_ERR_CORRUPT

lfs error:494: Corrupted dir pair at 3 2
virtual BluefruitFS::File LittleFS::open(const char*, uint8_t): 313: verify failed, error = LFS_ERR_CORRUPT
```

At that point, the only way I was able to recover FS functionality was to re-flash the bootloader and re-flash clearbonds which gave me: 

```
[CFG   ] SoftDevice's RAM requires: 0x20002530
lfs error:494: Corrupted dir pair at 0 1
lfs error:2222: Invalid superblock at 0 1
[IFLASH] Format internal file system
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS

----- Before -----

[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS

----- After  -----

```

After reflashing over the reformatted file system, it works fine, until the offending situation occurs and the problem happens again.

When the bonding information is wiped by reflashing the bootloader, I can unpair and reconnect in windows 10 and I get these messages when pairing:

```

[GAP   ] Conn Interval= 11.250000
[BLE   ] BLE_GATTC_EVT_CHAR_VAL_BY_UUID_READ_RSP : Conn Handle = 1
[BLE   ] BLE_GAP_EVT_SEC_PARAMS_REQUEST : Conn Handle = 1
[PRPH  ] Connected to 1 TOASTER
[BLE   ] BLE_GATTS_EVT_EXCHANGE_MTU_REQUEST : Conn Handle = 1
[GAP   ] ATT MTU is changed to 23
[BLE   ] BLE_GAP_EVT_CONN_SEC_UPDATE : Conn Handle = 1
[BOND  ] CCCD setting not found
[BLE   ] BLE_GAP_EVT_AUTH_STATUS : Conn Handle = 1
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[BLE   ] BLE_GATTC_EVT_CHAR_VAL_BY_UUID_READ_RSP : Conn Handle = 1
[BOND  ] Saved keys for "TOASTER" to file /adafruit/bond_prph/59d1 ( 90 bytes )
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[BLE   ] BLE_GATTS_EVT_WRITE : Conn Handle = 1
[BLE   ] BLE_GAP_EVT_CONN_PARAM_UPDATE : Conn Handle = 1
[GAP   ] Conn Interval= 11.250000
[BLE   ] BLE_GATTS_EVT_WRITE : Conn Handle = 1
[GATTS ] attr's cccd
000:  01 00                                           | ..

[BOND  ] Saved CCCD setting to file /adafruit/bond_prph/59d1 ( offset = 90, len = 44 bytes )
[BLE   ] BLE_GATTS_EVT_WRITE : Conn Handle = 1
[GATTS ] attr's cccd
000:  01 00                                           | ..

[BLE   ] BLE_GAP_EVT_CONN_PARAM_UPDATE : Conn Handle = 1
[GAP   ] Conn Interval= 56.250000
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[BLE   ] BLE_GATTS_EVT_WRITE : Conn Handle = 1
[GATTS ] attr's cccd
000:  01 00                                           | ..

[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[BOND  ] Saved CCCD setting to file /adafruit/bond_prph/59d1 ( offset = 90, len = 44 bytes )
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[BLE   ] BLE_GAP_EVT_CONN_PARAM_UPDATE : Conn Handle = 1
[GAP   ] Conn Interval= 11.250000
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[BOND  ] Saved CCCD setting to file /adafruit/bond_prph/59d1 ( offset = 90, len = 44 bytes )
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[BLE   ] BLE_GATTS_EVT_WRITE : Conn Handle = 1
[GATTS ] attr's value, uuid = 0x2A4D
000:  01                                              | .

[HID   ] Keyboard LED : 0x01
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
```

hathach, 2019-01-29:
begin() won't help, you need to call the format()

```
InternalFS.format(true);
```

true mean does low-level format.

jpconstantineau, 2019-02-02:
Great.  I'll keep an eye on this issue as I work on this over the weekend.

jeremypoulter, 2019-03-15:
FYI, I am investigating a similar issue. I am seeing LittleFS corruption when writing to a file from the BLE task, in the receive callback.

jeremypoulter, 2019-03-15:
So I resolved my issue by not writing from the bleuart receive callback and into the loop. Bit weird as there is no possibility of read/writes happening from different tasks. Does LittleFS use some task local storage (if that is even possible in FreeRTOS)?

hathach, 2019-03-18:
`LittleFS` so far is only used by bonding.cpp module when pairing and/or cccd is written by client. However, the actual flash writing is deferred via ada_callback() instead of calling within ble task to reduce latency. Let's me know if you guys need more details on this.

matthewswilson, 2019-03-21:
I have been experiencing similar random issues with the HID keyboard.  Most of the issues I can't recreate because they are random.  For example, the first paring to the iPhone works great but if I turn phone or device off the keyboard never comes back up but shows connected.  Sometimes it won't reconnect.  Usually the only way to fix it is to remove device then repair or refresh the boot loader.  I'm taken some of the ideas above and building a reset feature.  My reset logic is triggered on a sequence of button presses.  I provided my code snippet below for when the reset is triggered.  I'm getting an error LFS_ERR_NOENT.  What does this mean when I'm clearing the bond data?

LOG_LV2("RESET","Formatting Filesystem");
InternalFS.begin();
InternalFS.format(true);
LOG_LV2("RESET","Clearing Bond Data");
Bluefruit.clearBonds();
//Bluefruit.Central.clearBonds();

[RESET ] Formatting Filesystem
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
......
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[RESET ] Clearing Bond Data
virtual bool LittleFS::rmdir_r(const char*): 373: verify failed, error = LFS_ERR_NOENT
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
......
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS

P.S. I purchased a bluetooth keyboard from Amazon to see how it would act.  It works extremely well.  In fact I can have the Bluetooth keyboard connected and the iPhone keyboard up at the same time and use both.  I'm really trying to dig in and understand what happens during the bonding stage between the device and phone.  How is the Amazon keyboard presenting itself in a way that's is different the Bluefruit?  Just some questions I'm trying to understand to help solve these issues.

Thanks...

shaoyuan1943, 2019-07-11:
@hathach Hi, today I got same error, the serial log:
`lfs error:494: Corrupted dir pair at 34 33`
`22:35:54.643 -> virtual BluefruitFS::File LittleFS::open(const char*, uint8_t): 267: verify failed, error = LFS_ERR_CORRUPT`

Then I re-flash bootloader and rebuild my code, above error is gone, but there has new error:
`01:45:01.850 -> bool BLEAdvertising::_start(uint16_t, uint16_t): 364: verify failed, error = NRF_ERROR_INVALID_STATE`

I don't understand that this problem suddenly appears, my code has not been modified and that work fine yesterday. I'm trying use this code to resolve the error but it not work, error always:
`InternalFS.begin();
InternalFS.format();`

My work has been interrupted by this problem. I am looking for a way to solve this problem. Do you know how to solve this problem?

Thank you...

PR #397 (2020-01-06): Serialize lfs
Fix #350
Fix #325
Fix #227
Fix #222
Fix #393

@hathach -- This PR supercedes #372 (Guard against race conditions in Flash FS cache), as #72 was not really providing a proper fix of the underlying problem:

LFS itself is [not safe](https://github.com/ARMmbed/littlefs/issues/352) to call simultaneously from multiple contexts.

This PR solves this by wrapping the InternalFS API with a non-recursive mutex, as recommended.

I realize the PR uses newly declared wrapper functions.  I tried to avoid this, but the changes got messy quite quickly, mostly due to some of the macros used to v [... cut]
closed 2020-01-06 by hathach