adafruit/Adafruit_nRF52_Arduino #224: nRF52840 Express Stuck on Upgrading target on...

czuvich, 2019-02-04:
I'm trying to upload some of the example sketches onto the nRF52840 Express Feather chip; however, I am getting error messages uploading.  I'm running on MacOS, bootloader 0.2.6, Arduino IDE 1.8.8.  

Sometimes I am able to upload the sketch; however, once I get this error I get hosed for awhile (at some point the Serial port opens up).  I've tried uploading the same sketches on Windows as well, and it seems to be more stable; however, I still get this issue.  Any help is greatly appreciated.

```
Upgrading target on /dev/cu.usbmodem1414101 with DFU package /private/var/folders/_d/ksp5fdyx5h1flr8s6l6rvgj00000gn/T/arduino_build_728309/neopixel.ino.zip. Flow control is disabled, Single bank, Touch disabled
Timed out waiting for acknowledgement from device.

Failed to upgrade target. Error is: No data received on serial port. Not able to proceed.
Traceback (most recent call last):
  File "nordicsemi/__main__.py", line 294, in serial
  File "nordicsemi/dfu/dfu.py", line 235, in dfu_send_images
  File "nordicsemi/dfu/dfu.py", line 200, in _dfu_send_image
  File "nordicsemi/dfu/dfu_transport_serial.py", line 179, in send_start_dfu
  File "nordicsemi/dfu/dfu_transport_serial.py", line 243, in send_packet
  File "nordicsemi/dfu/dfu_transport_serial.py", line 282, in get_ack_nr
nordicsemi.exceptions.NordicSemiException: No data received on serial port. Not able to proceed.

Possible causes:
- Selected Bootloader version does not match the one on Bluefruit device.
    Please upgrade the Bootloader or select correct version in Tools->Bootloader.
- Baud rate must be 115200, Flow control must be off.
- Target is not in DFU mode. Ground DFU pin and RESET and release both to enter DFU mode.
```

hathach, 2019-02-11:
thanks for reporting the issue, it is annoying, seems to be driver related issue. In Linux, i have to disable modem manager module since it interfere with serial port. Not sure if that is needed for macos, I will try to test more with macos. Btw, when the issue occurs, power off and on may fix the issue (like double reset). Sorry for the inconvenience.

czuvich, 2019-02-11:
Thanks @hathach.  You guys may want to consider mentioning this issue somewhere that others can easily see it depending on how long it takes to fix it.  I spent hours trying to figure out what was going on.  As you know it literally halts all development effort until it gets resolved.  I can only imagine how frustrating it could be to someone who isn't as "programmer" savvy.

hathach, 2019-05-02:
@czuvich did the resolve the issue by disable the modemman or using other way ?

czuvich, 2019-05-02:
@hathach I'm not familiar with modemman (I'm on Mac).  I had to always double-tap the reset button.  I can try disabling it if you have some instructions (assuming it won't have an impact on my mac).

hathach, 2019-05-03:
I guess it may not be related to `modemman` afterall otherwise, lots of people will be affected. Can you tell which version of your macOS, also maybe try to upgrade bootloader to later version `0.2.9` may help.

czuvich, 2019-05-03:
I'm using macOS Mojave (10.14.3), and I also run Windows 10.  On both OS, I still see the same issue with 0.2.9.  I've been developing against the nRF52840 Express for the last several weeks, and I have not noticed any changes in the issue behavior.  

It seems to be directly related to when a BLE sketch is uploaded.  If you're not enabling the Soft Device BLE API's (Bluefruit.begin), then I have no issues uploading sketches without a double tap.  I hope that helps.

hathach, 2019-05-03:
Thanks for your infor,  csn you also tell us which version of bsp you are using. I woukd recommend 0.9.3 or 0.10.1, thougb the later include lots of breaking changes

czuvich, 2019-05-03:
I'm currently using 0.10.1.  I did update to the new changes when it was released (support for multiple connections right?).

hathach, 2019-05-03:
Yeah, you are already at latest, I currently have no clues why it occurs since I couldn't reproduce it. Let's leave it as it is, we will do more tests with more machine later on or wait to see other user also got this to find the common ground.

czuvich, 2019-05-03:
Ok.  I have purchased about a dozen or so of these nRF chips, and I get the same behavior on all of them (on both my Windows and Mac).  I've also seen other posts about it on the forums.  Might be my cable... dunno.  I'll keep you posted if I discover anything new.

TechplexEngineer, 2019-05-19:
Double reset doesn't work for me. On Linux or OSX.

schizobovine, 2019-08-23:
Same. Device is near bricked at this point and about to break out JTAG just to flash the thing. It's not modemmanager because I removed that before even installing arduino. I also disabled automounting in case that had something to do with it (since that also seems to trigger on restart).

Also, double reset does NOT work at all for me. It just goes to the green "running" mode but doesn't accept any new sketches nor show up on the Bluefruit app.

lorenzfischer, 2020-11-18:
Not sure if this similar to what you guys were experiencing, but in case someone else comes by here, I was able to resolve the above issue (on a Mac running Catalina) by updating the Bootloader of the Feather using [these](https://learn.adafruit.com/bluefruit-nrf52-feather-learning-guide/updating-the-bootloader) instructions.

italoraony, 2021-01-27:
@hathach and @lorenzfischer - Thank you so much! 

The double reset + updated Bootloader worked perfectly!

closed 2022-01-11 by czuvich