adafruit/Adafruit_nRF52_Arduino #225: BSP won't install on Raspberry Pi

mml, 2019-02-05:
According to the IDE, this is because "Tool nrfjprog is not available for your operating system."  I think this is because nrfjprog is not distributed for ARM.  But AFAICT, I would only need that tool if I wanted to flash bootloader.  In order to compile and upload sketches, I only need adafruit-nrfutil.  (I already updated the bootloader using my Mac.)

I don't know enough about the BSP spec, but one idea was to just hack it and tell Arduino IDE that on ARM, `/bin/false` is this program.

- Board: Feather nRF52832
- OS: Raspbian
- Arduino IDE: 1.8.8

hathach, 2019-02-11:
nrfjprog is not required to use featehr nrf52, you only need adafruit-nrfutil to perform serial DFU. You can install adafruit-nrfutil via pip package I guess

https://learn.adafruit.com/bluefruit-nrf52-feather-learning-guide/arduino-bsp-setup#2-linux-only-adafruit-nrfutil-tool-installation-4-12

rolandoam, 2020-11-27:
had the same issue. I duplicated the adafruit package json and added a dummy aarch64 entry, replaced the adafruit package index with my own and that allowed me to install this on a rpi4

egnor, 2021-10-02:
Gentle nudge on this! Can we somehow make this dependency optional, at least on this architecture?

hathach, 2021-10-14:
just double checked, Nordic has release nrfjprog for ARM with latest version. I will include this as part of the next release and post update here.

closed 2021-11-26 by hathach

hathach, 2021-11-26:
release 1.2.0 should be installable on rpi os