adafruit/Adafruit_nRF52_Arduino #226: ExternalFS nRF52840 Fail

czuvich, 2019-02-09:
I'm using the nRF52840 Express in the Arduino IDE, and I'm trying to run the External_ListFiles sketch. I want to be able to write data to the QSPI flash chip (not the internal flash). When I try to run the example, it says the following:

BluefruitFS::File FatFS::_open_dir(const char*): 225: verify failed, error = FR_DISK_ERR

Is there something I'm doing wrong? I'm using 0.2.6 bootloader (however, I just got my SWD connector, so I'm going to try updating to 0.2.7 today).

czuvich, 2019-02-09:
I also wanted to mention while I'm on here, I think there are issues with the InternalFS.  I am getting corrupted blocks somewhat often, and I end up having to format it to fix it.  I was using originally declaring a File instance inside of each method that needed; however, I have since changed to a single instance of File(InternalFS).

I'll keep you posted if I continue seeing the corrupted block issues with a singleton instance of File(InternalFS).

hathach, 2019-02-11:
currently, feather nrf52840 doesn't support readding/writing to qspi flash yet, due to an issue with qspi driver 
https://github.com/adafruit/Adafruit_nRF52_Arduino/issues/214
We will try to fix the qspi driver later on and include it in later release along with USB MSC interface.

For InternalFS, it is a separated issue, please open another issue for it. There is other issue related to InternalFS corrupted fs as well
https://github.com/adafruit/Adafruit_nRF52_Arduino/issues/222

czuvich, 2019-02-11:
Thanks @hathach.  I'll post a separate issue.

hathach, 2019-06-19:
ExternalFS is supported by SdFat in https://github.com/adafruit/Adafruit_SPIFlash
check out msc_external_flash example for usage https://github.com/adafruit/Adafruit_TinyUSB_Arduino

closed 2019-06-19 by hathach