adafruit/Adafruit_nRF52_Arduino #227: InternalFS Corrupt Blocks

czuvich, 2019-02-11:
Board: nRF52840 Express
MacOS
Bootloader 0.2.6
Arduino IDE 1.8.8

I'm trying to write a string to the InternalFS, and I get the following error sporadically.
```
12:59:53.579 -> lfs error:494: Corrupted dir pair at 0 1
12:59:53.579 -> lfs error:2222: Invalid superblock at 0 1
12:59:53.579 -> [IFLASH] Format internal file system
```
The only way to fix it is to InternalFS.format(true).  I can't quite pinpoint what is causing it, so I don't think posting code will help.  Having said that, I did some research and discovered the following which may/may not help:

1) InternalFS File.write(buffer, size) does not overwrite the file. It appends to it.  I incremented a counter every 1s and read/write it to the same file and got the following output:
```
1
12
123
```
Instead, I would expect the following:
```
1
2
3
```
Which takes me to #2. 

2)  In order to overwrite the file contents, I remove the file and create a new one. Removing a file takes a VERY long time (a few seconds).  I found the following article from Nordic that describes how internal flash is slow to remove:

https://devzone.nordicsemi.com/f/nordic-q-a/20197/nrf52-flash-memory-vs-eeprom

It could be that since I'm running BLE and removing files, RTOS is switching tasks which is causing issues?  Anyways, right now I've created an RTOS queue to serialize my writes, and/or maybe I need to add a mutex.  Just thought I'd post my findings here.

hathach, 2019-02-15:
Thank you very much for posting the issue, there seem to be an issue with current internal filesystem #222 . Since it is dynamic and timing sensitive, which is hard to reproduce, we will investigate this later. 

The internal FS uses https://github.com/ARMmbed/littlefs you can check out their implementation/design for how FS works. Though, I would not try to alternate their design, we should only focus on flash vs sd relation and why it fails to write.

jeremypoulter, 2019-03-13:
For 1) this is the same issue as I reported in #234, seems like we have Arduino to thank for a non-standard API ;-)

hathach, 2019-03-14:
@jeremypoulter  @czuvich Ah thanks, I just realized I skim too fast and only think of the corruption issue. Yes the Arduino API FILE_WRITE is always in append mode. If you want overwrite, you could either remove or open then seek(0) to the beginning of the file and start writing.

PR #397 (2020-01-06): Serialize lfs
Fix #350
Fix #325
Fix #227
Fix #222
Fix #393

@hathach -- This PR supercedes #372 (Guard against race conditions in Flash FS cache), as #72 was not really providing a proper fix of the underlying problem:

LFS itself is [not safe](https://github.com/ARMmbed/littlefs/issues/352) to call simultaneously from multiple contexts.

This PR solves this by wrapping the InternalFS API with a non-recursive mutex, as recommended.

I realize the PR uses newly declared wrapper functions.  I tried to avoid this, but the changes got messy quite quickly, mostly due to some of the macros used to v [... cut]
closed 2020-01-06 by hathach