adafruit/Adafruit_nRF52_Arduino #228: Please add BLEHID gamepad support and example

uhrheber, 2019-02-13:
I want to use a nRF52840 as a BLE HID gamepad, but I saw, that the gamepad descriptor is commented out in BLEHidAdafruit.cpp. Is there something wrong with the code?

If so, what is needed to fix it?

Also, does somebody have an example for a gamepad?

dhalbert, 2019-02-13:
I don't know what's wrong with the BLE one, if anything, but here is a USB HID gamepad descriptor, used in CircuitPython, which could be transcribed into the BLE version: https://github.com/adafruit/circuitpython/blob/master/tools/hid_report_descriptors.py

This is recognized by Windows 10, MacOS, and Linux, in some fashion, without installing extra drivers. How useful it is, is another question. Gamepads are often highly specific for specific applications, and drivers or game consoles expect only certain descriptors.

I'm a little suspicious of the ranges in the joystick values in the BLE HID one: `0xFF` instead of a true negative value. Also it's only a a 2-bit field.

dhalbert, 2019-02-13:
BLE HID (including maybe gamepad) for CircuitPython is on our shorter-term list, but as usual, we have no specific schedule. I haven't started it yet.

hathach, 2019-02-14:
#36 Last time I tested, iOS doesn't really like gamepad as hid service (keyboard and mouse work fine). I will again later on.

uhrheber, 2019-02-14:
What about classic bluetooth?
The talk is all about BLE nowadays, but can I also implement a classic bluetooth gamepad using the nRF52 Arduino BSP?

dhalbert, 2019-02-14:
The nRF52840 doesn't support classic Bluetooth, and I believe there is no chip from Nordic that supports classic Bluetooth (BR/EDR).

hathach, 2019-02-14:
Like @DanHalbert said, this repo only support ble, no classic support. Please be patient, we will give gamepad another try later on, either having it as part of current adafruit hid service or separated service.

uhrheber, 2019-02-22:
@hathach Not only iOS, I also noticed some weird behaviour in Windows 10, where USB and Bluetooth HID devices seem to interfere with each other. I had to remove a custom USB HID keyboard device to get the Bluetooth HID gamepad  working. I don't know, whether this was my fault or a bug in Windows, but it definitely shouldn't happen.

hathach, 2019-02-22:
Yeah, we need time to test it with major OS. Been too busy lately with other features :(

uhrheber, 2019-02-23:
Hey, nobody's blaming you. Bluetooth is disappointingly complicated. Even pros try to avoid the effort and buy modules with ready-made firmware instead.

I didn't even manage to read the first chapter of my BLE book yet.

hathach, 2019-07-10:
#36

jameshallam, 2019-08-14:
I'm going to follow this issue - we are starting to use the nrf52840 quite extensively, and Arduino BLE gamepad support for this board would be hugely beneficial. Thanks for considering it!

mne-4d, 2019-09-24:
Hi,

I follow this issue too, it would be really useful to simply add a HID gamepad to some projects I'm working on

unixunion, 2020-08-30:
I found this PR, https://github.com/adafruit/Adafruit_BluefruitLE_nRF51/pull/37/files, any chance of a merge any time soon_