adafruit/Adafruit_nRF52_Arduino #233: Add Support for CMSIS-DAP (OPENOCD)

ericlangel, 2019-03-04:
Hello,

is it possible to add the CMSIS-DAP programmer?

i already try to implement it, but the flashed Application didn't work. The Bootloader LED flashed all the time.

Here are my changes:
platform.txt

enable .bin Output an add:

## OpenOCD tool

#tools.openocd.path={runtime.tools.openocd.path}
tools.openocd.path={runtime.tools.openocd-0.10.0-dev-nrf5.path}
#tools.openocd.path={runtime.platform.path}/tools/openocd
tools.openocd.cmd.macosx=bin/openocd
tools.openocd.cmd.linux=bin/openocd
tools.openocd.cmd.windows=bin/openocd.exe

tools.openocd.upload.params.verbose=-d1
tools.openocd.upload.params.quiet=-d0
tools.openocd.upload.pattern="{path}/{cmd}" {upload.params.verbose} -s "{path}/scripts/" -f "{path}/scripts/interface/{upload.interface}.cfg" -f "{path}/scripts/target/{upload.target}.cfg" -c "init;reset halt;program {{{build.path}/{build.project_name}.bin}} verify reset exit 0x26000"

tools.openocd.program.params.verbose=-d2
tools.openocd.program.params.quiet=-d0
tools.openocd.program.pattern=

tools.openocd.erase.params.verbose=-d3
tools.openocd.erase.params.quiet=-d0
tools.openocd.erase.pattern=

tools.openocd.bootloader.params.verbose=-d1
tools.openocd.bootloader.params.quiet=-d0
tools.openocd.bootloader.pattern="{path}/{cmd}" {bootloader.verbose} -s "{path}/scripts/" -f "{path}/scripts/interface/{upload.interface}.cfg" -f "{path}/scripts/target/{upload.target}.cfg" -c "init;reset halt;nrf51 mass_erase; program {{{runtime.platform.path}/bootloaders/{bootloader.bootloader_file}}} verify reset exit"

boards.txt

#**********************************************
# Feather nRF52832 Board (DAP)
#**********************************************
feather52832D.name=Adafruit Bluefruit nRF52832 Feather (DAP)
feather52832D.bootloader.tool=bootburn

# Upload
feather52832D.upload.tool=openocd
feather52832D.upload.protocol=
feather52832D.upload.use_1200bps_touch=false
feather52832D.upload.wait_for_upload_port=false
feather52832D.upload.interface=cmsis-dap
feather52832D.upload.target=nrf52
feather52832D.upload.maximum_size=524288
feather52832D.upload.maximum_data_size=65536
feather52832D.upload.speed=115200
feather52832D.upload.disable_flushing=false

hathach, 2019-03-04:
Since we don't use nor recommend user to use openocd/cmsisdap. We won't put effort to implement or maintain this. However, we are happy to merge PR that have this working.

hathach, 2019-04-08:
closed due to lack of user interaction

closed 2019-04-08 by hathach