adafruit/Adafruit_nRF52_Arduino #234: LittleFS write behaviour

jeremypoulter, 2019-03-09:
This [commit](https://github.com/Adafruit/Adafruit_nRF52_Arduino/commit/7a23c671fd4cac43c3925d0390e458693b3fb370) looks to have removed append support and made Write always append. This is very non-standard and means to overwrite a file you have to first delete it. Can this be reverted or are you able to provide some more information about why the is the 'correct' behaviour?

hathach, 2019-03-11:
@jeremypoulter you got an eagle eye, but this is on-purpose, I did put an move to end when open with FILE_WRITE mode (write is always append since Arduino only have FILE_READ & FILE_WRITE https://www.arduino.cc/en/Reference/SDopen )

https://github.com/Adafruit/Adafruit_nRF52_Arduino/commit/7a23c671fd4cac43c3925d0390e458693b3fb370#diff-76f647aab692ab28f74e3ab20afd6c8eR234

The reason is LFS Append flag **force move to the end on every file write()** making it hard to overwrite a middle of file hard with multiple file and not really what a normal user could expect

https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/libraries/FileSystem/src/littlefs/lfs.h#L85
https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/libraries/FileSystem/src/littlefs/lfs.c#L1643

Let's me know if you still have comment on this.

jeremypoulter, 2019-03-11:
Fair enough, sometimes the Arduino folks make some dumb APIs ;-)

closed 2019-03-11 by jeremypoulter

hathach, 2019-08-29:
> This is very unusual. Couldn't there be a separate file mode for appends, like FILE_A_WRITE, and keep FILE_O_WRITE for overwrites?
> 
> [Adafruit_LittleFS_File.h](https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/libraries/Adafruit_LittleFS/src/Adafruit_LittleFS_File.h)
> 
> ```
> // avoid conflict with other FileSystem FILE_READ/FILE_WRITE
> enum
> {
>   FILE_O_READ = 0,
>   FILE_O_WRITE = 1,
> };
> ```

It is unusual but  follows Arduino API design. I would suggest to just leave it as READ/WRITE, if you could submit an PR for `FILE_O_OVERWRITE` I will review and merge