adafruit/Adafruit_nRF52_Arduino #235: LittleFS format memory leak

jeremypoulter, 2019-03-09:
`lfs_format` internally calls `lfs_init` so if you call `LittleFS.format` after `LittleFS.begin` (as would be expected in the Arduino world) it will reallocate the leak the buffers allocated as a result of calling `LittleFS.begin`. 

Is this expected behaviour or a bug.

Note: I have resolved this locally by doing the following;
```c++
bool LittleFS::format (bool eraseall)
{
  if ( eraseall ) {
    _flash_erase_all();
  }
  if(_mounted) {
    VERIFY_LFS(lfs_unmount(&_lfs), false);
  }
  VERIFY_LFS(lfs_format(&_lfs, &_lfs_cfg), false);
  VERIFY_LFS(lfs_mount(&_lfs, &_lfs_cfg), false);

  _mounted = true;

  return true;
}
```

jeremypoulter, 2019-03-09:
(I will raise a PR if you think this is a bug)

hathach, 2019-03-11:
@jeremypoulter wow, such a nice catch, it is indeed a bug. Even me call Format() after Begin() in the example sketch as well 
https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/libraries/FileSystem/examples/Internal_Format/Internal_Format.ino

Please submit a PR for this if you have time, the code have `_begun`, we could use that in place of your `_mounted` above. Actually, we should have `_begun` reflect the mount state instead of blindly true as right now. Don't worry about it for your PR, I will update it after the merge. Thank you very much.

PR #238 (2019-03-12): Unmount the FS before formatting
Fixes #235
closed 2019-03-12 by hathach