adafruit/Adafruit_nRF52_Arduino #241: I2C (Wire) hangs when no device is attached.

beegee-tokyo, 2019-03-12:
- Adafruit Feather nRF52 Bluefruit LE - nRF52832 board
  - Adafruit_nRF52_Arduino repo version 0.9.3
  - Bootloader/Softdevice version 0.2.6 S132 6.1.1
  - Arduino IDE 1.8.5 on Windows 7 64bit

I have in my SW a simple I2C scan function to find device connected to the board. This function worked perfect before last update of this repo. If I found the devices I am looking for and the flags were set. If I didn't find the devices, the flags were not set. After that my software just continued and used the sensor that were found or gave an error message if no devices were found.
Since the last update of the repo, if there are no devices connected, the SW just hangs in the ```i2cWire2.beginTransmission(address);``` function and never returns from it.

The weird thing is that it happens only on the I2C bus on pins 25 and 26. The second I2C bus on pins 15 and 16 works as expected.
And this behavior can be replicated on all my nRF52 feather boards that I have.

Any idea why? I didn't find commits that seem to change something in the Wire library.

**_Scan function:_**
```CPP
/** SDA pin for first I2C */
#define SDA0 PIN_WIRE_SDA
/** SCL pin for first I2C */
#define SCL0 PIN_WIRE_SCL
/** SDA pin for second I2C */
#define SDA1 15
/** SCL pin for second I2C */
#define SCL1 16

/** First sensor Wire class */
TwoWire i2cWire1 = TwoWire(NRF_TWIM0, NRF_TWIS0, SPIM0_SPIS0_TWIM0_TWIS0_SPI0_TWI0_IRQn, SDA0, SCL0);
/** Second sensor Wire class */
TwoWire i2cWire2 = TwoWire(NRF_TWIM1, NRF_TWIS1, SPIM1_SPIS1_TWIM1_TWIS1_SPI1_TWI1_IRQn, SDA1, SCL1);

/** Sensor type */
/** bit 0 = Sensor 1 = MAX30105 */
/** bit 1 = Sensor 2 = MAX30105 */
/** bit 2 = Sensor 1 = Flex sensor */
/** bit 3 = Sensor 2 = Flex sensor */
byte sensorType = 0;

/** Flag if sensor 1 is a MAX sensor */
bool sens1Max = false;
/** Flag if sensor 1 is a Flex sensor */
bool sens1Flex = false;
/** Flag if sensor 2 is a MAX sensor */
bool sens2Max = false;
/** Flag if sensor 2 is a Flex sensor */
bool sens2Flex = false;

bool initSensors(void)
{
	Serial.println("Initializing sensors");

	// Scan the I2C interfaces for devices
	byte error;
	Serial.println("Start I2C #1");
	i2cWire1.begin();
	i2cWire1.setClock(I2C_SPEED_STANDARD);
	Serial.println("Scan I2C bus #1");
	for (byte address = 1; address < 127; address++)
	{
		Serial.print("Trying address: 0x");
		Serial.println(address);
		i2cWire1.beginTransmission(address);
		error = i2cWire1.endTransmission();
		if (error == 0)
		{
			Serial.print("Found device on I2C #1 on address ");
			Serial.println(address, HEX);
			if (address == 0x57)
			{
				sensorType |= 0x01;
				sens1Max = true;
				Serial.println("MAX30105 sensor found on I2C 1");
[... 312 lines cut ...]
Trying address: 0x116
Trying address: 0x117
Trying address: 0x118
Trying address: 0x119
Trying address: 0x120
Trying address: 0x121
Trying address: 0x122
Trying address: 0x123
Trying address: 0x124
Trying address: 0x125
Trying address: 0x126
Initializing MAX sensor on I2C 1
MAX30102 init on I2C 1 successfull!
Switching off the MAX sensor on I2C 1
ID: PPG-04E1E1CD2CACC007
---------------------------

BLE advertising started
MAX30102 or Flex sensors found, start reading the sensors
---------------------------

Starting battery timer
Starting Sensor processing task
Starting Sensor processing task success
```

hathach, 2019-03-13:
Thanks for raising the issue, Can you specify the previous working version with your sketch. If possible could you give us the minimal version of your sketch that causes the issue, that will help me to quickly reproduce and troubleshoot.

beegee-tokyo, 2019-03-13:
@hathach 
I am not sure with which commit the problem started. I didn't update the repo for a long time, just switched maybe a week ago. I remember that when it was working there was still an older bootloader/softdevice in the repo, because I had to update the bootloader/softdevice on the nRF52 after updating the repo.

For the minimal sketch, here is a more reduced version of it:
```
#include <bluefruit.h>
#include <Arduino.h>
#include <Wire.h>

/** SDA pin for first I2C */
#define SDA0 PIN_WIRE_SDA // GPIO 25
/** SCL pin for first I2C */
#define SCL0 PIN_WIRE_SCL // GPIO 26

/** First sensor Wire class */
TwoWire i2cWire1 = TwoWire(NRF_TWIM0, NRF_TWIS0, SPIM0_SPIS0_TWIM0_TWIS0_SPI0_TWI0_IRQn, SDA0, SCL0);

void setup(void)
{
	Serial.begin(115200);

	Serial.println("Initializing sensors");

	// Scan the I2C interfaces for devices
	byte error;
	Serial.println("Start I2C #1");
	i2cWire1.begin();
	i2cWire1.setClock(I2C_SPEED_STANDARD);
	Serial.println("Scan I2C bus #1");
	for (byte address = 1; address < 127; address++)
	{
		Serial.print("Trying address: 0x");
		Serial.println(address,HEX);
		i2cWire1.beginTransmission(address);
		error = i2cWire1.endTransmission();
		if (error == 0)
		{
			Serial.print("Found device on I2C #1 on address ");
			Serial.println(address, HEX);
		}
	}
	i2cWire1.end();
｝
```
The strange thing is that if I change the pins used for SDA and SCL it seems to work. On pins 25 and 26 is nothing connected than the sensor, the pins are open if the sensor is not attached. The I2C pullup resistors are on the sensor board, but I tried to add a pullup to the SDA pin when no sensor is connected, but it still hangs.
It too doesn't make any difference if I initialize the I2C with 
```TwoWire i2cWire1 = TwoWire(NRF_TWIM0, NRF_TWIS0, SPIM0_SPIS0_TWIM0_TWIS0_SPI0_TWI0_IRQn, SDA0, SCL0);```
or
```TwoWire i2cWire1 = TwoWire(NRF_TWIM1, NRF_TWIS1, SPIM1_SPIS1_TWIM1_TWIS1_SPI1_TWI1_IRQn, SDA0, SCL0);```

beegee-tokyo, 2019-03-13:
Update:
When I change the SCL pin to any other available pin than 26 the scan works. Only if I use GPIO26 the scan hangs on the first address.
Tested with the minimal sketch I posted above.

beegee-tokyo, 2019-03-21:
@hathach 
any findings?
Will check tomorrow if swapping default SDA and SCL {GPIO 25 and 26) works as well.

hathach, 2019-03-22:
Unfortunately, I haven't got time to work on this yet

beegee-tokyo, 2019-03-22:
Just in case you find time, another weird thing.
If I swap SDA from GPIO 25 (default) and SCL from GPIO 26 (default) to SDA on GPIO 26 and SCL on GPIO25 it **_hangs as well_** if no sensor is attached.

Points into the direction that it has something todo with GPIO 26. But I have no idea why.

ladyada, 2019-03-22:
bee, do you have i2c pullups on your board, even when no sensor is attached?

beegee-tokyo, 2019-03-23:
@ladyada 
The I2C pullups are on the sensor board, so when the sensor is not attached the I2C lines are open.

I tried just now to put 470Ohm resistors on the nRF52 board, so that both SDA and SCL are pulled up if no sensor is attached and then the I2C scan works. :1st_place_medal: 

### But
**_Why does the I2C scan work on the second I2C (GPIO 15 and GPIO 16) without pull-ups and without sensor attached?
And why does the I2C scan work on the first I2C if I use any other pin than GPIO 26?_**

Using the pull-ups will be a solution for the final product we are working on, but right now we are using proto boards where the pull-ups are on the sensor board and not on the nRF52 Feather board.

Will discuss with our HW guy how to solve this.

ladyada, 2019-03-23:
the i2c spec says you *must* have pullups when using i2c, other behavior is not guaranteed. so add 10K pullups :)

beegee-tokyo, 2019-03-23:
**`TwoWire::begin(void)`** sets the SDA and SCL lines to use internal pull-up. Shouldn't that be sufficient to get the I2C to work if no device is attached:
Line 52...62 Wire_nRF52.cpp
```
  *pincfg_reg(_uc_pinSCL) = ((uint32_t)GPIO_PIN_CNF_DIR_Input        << GPIO_PIN_CNF_DIR_Pos)
                           | ((uint32_t)GPIO_PIN_CNF_INPUT_Connect    << GPIO_PIN_CNF_INPUT_Pos)
                           | ((uint32_t)GPIO_PIN_CNF_PULL_Pullup      << GPIO_PIN_CNF_PULL_Pos)
                           | ((uint32_t)GPIO_PIN_CNF_DRIVE_S0D1       << GPIO_PIN_CNF_DRIVE_Pos)
                           | ((uint32_t)GPIO_PIN_CNF_SENSE_Disabled   << GPIO_PIN_CNF_SENSE_Pos);

  *pincfg_reg(_uc_pinSDA) = ((uint32_t)GPIO_PIN_CNF_DIR_Input        << GPIO_PIN_CNF_DIR_Pos)
                           | ((uint32_t)GPIO_PIN_CNF_INPUT_Connect    << GPIO_PIN_CNF_INPUT_Pos)
                           | ((uint32_t)GPIO_PIN_CNF_PULL_Pullup      << GPIO_PIN_CNF_PULL_Pos)
                           | ((uint32_t)GPIO_PIN_CNF_DRIVE_S0D1       << GPIO_PIN_CNF_DRIVE_Pos)
                           | ((uint32_t)GPIO_PIN_CNF_SENSE_Disabled   << GPIO_PIN_CNF_SENSE_Pos);
```

ladyada, 2019-03-23:
the pullups are very weak, dont rely on them

beegee-tokyo, 2019-03-23:
Some test:
SDA0 is always configured on P0.25 as input with internal pullup and without external pullup
SCL0 is configured to use P0.26 or as P0.27 with and without external pullup.
All tests were done without any device connected to the I2C lines.
First three tests were done with above simple I2C scan function.
Last test was done with as simple code that just toggles SCL line.

I configured I2C to use P0.25 as SDA and **_P0.26_** as SCL
First two test SCL (blue) is **_without_** external pullup and SDA (yellow) is **_without_** external pullup
[image]
The SCL line does not reach 3.0 V despite the fact that it should be configured as an output. On top it only goes up once and never toggles again.
SDA however goes up to the expected 3.3V

Second test SCL (blue) is **_with_** external pullup and SDA (yellow) is **_without_** external pullup
[image]
Now SCL reaches the 3.3V and is acting as a clock. SDA line is same again (3.3V)

Next test I configured I2C to use P0.25 as SDA and **_P0.27_** as SCL
Here is the behaviour **_without_** any pullups on SDA nor SCL
[image]
As you can see the SCL line now reaches **_without_** external pullup the 3.3V 

At last, I wrote a quick test routine that uses P0.26 as output and P0.25 as input.
Both are open and **_without_** external pullups and P0.25 is defined as INPUT_PULLUP.
```
/** SDA pin for first I2C */
#define SDA0 PIN_WIRE_SDA // = 25
/** SCL pin for first I2C */
#define SCL0 PIN_WIRE_SCL // = 26

	pinMode(SDA0, INPUT_PULLUP);
	pinMode(SCL0, OUTPUT);
	digitalWrite(SCL0, HIGH);
	while (1)
	{
		digitalWrite(SCL0, !digitalRead(SCL0));
		delay_ns(20);
	}
```
Now the SCL line behaves as expected.
[image]

So there is no problem on P0.26 if used as normal GPIO with internal pullup enabled and driven by SW to act as a clock signal.

Conclusion:
Somewhere in the depth of nRF52 code or Adafruit nRF52 Arduino code is something strange happening with P0.26 and it is not happening if SCL is put on any other pin than P0.26

beegee-tokyo, 2019-04-04:
Closing this issue because it might not be a SW problem, but a HW problem on the Adafruit nRF52 Feather board.

### **_Finally got one of my own proto-boards up and running:_**
[image]

**_On this board there is no problem with the I2C if no sensor is attached. So I guess it must be a problem on the nRF52 Feather board._**

Testcode:
```CPP
#include <bluefruit.h>
#include <Arduino.h>
#include <Wire.h>

/** SDA pin for first I2C */
#define SDA0 PIN_WIRE_SDA // GPIO 25
/** SCL pin for first I2C */
#define SCL0 PIN_WIRE_SCL // GPIO 26

#define I2C_SPEED_STANDARD        100000
#define I2C_SPEED_FAST            400000

/** First sensor Wire class */
TwoWire i2cWire1 = TwoWire(NRF_TWIM0, NRF_TWIS0, SPIM0_SPIS0_TWIM0_TWIS0_SPI0_TWI0_IRQn, SDA0, SCL0);

void setup(void)
{
  Serial.begin(115200);

  Serial.println("Initializing sensors");

  // Scan the I2C interfaces for devices
  byte error;
  Serial.println("Start I2C #1");
  i2cWire1.begin();
  i2cWire1.setClock(I2C_SPEED_STANDARD);
  Serial.println("Scan I2C bus #1");
  for (byte address = 1; address < 127; address++)
  {
    Serial.print("Trying address: 0x");
    Serial.println(address,HEX);
    i2cWire1.beginTransmission(address);
    error = i2cWire1.endTransmission();
    if (error == 0)
    {
      Serial.print("Found device on I2C #1 on address ");
      Serial.println(address, HEX);
    }
  }
  i2cWire1.end();
}

void loop(void)
{
}```
Debug output:
```
Initializing sensors
Start I2C #1
Scan I2C bus #1
Trying address: 0x1
Trying address: 0x2
Trying address: 0x3
Trying address: 0x4
Trying address: 0x5
Trying address: 0x6
Trying address: 0x7
Trying address: 0x8
Trying address: 0x9
Trying address: 0xA
Trying address: 0xB
Trying address: 0xC
Trying address: 0xD
Trying address: 0xE
Trying address: 0xF
Trying address: 0x10
Trying address: 0x11
Trying address: 0x12
Trying address: 0x13
Trying address: 0x14
Trying address: 0x15
Trying address: 0x16
Trying address: 0x17
Trying address: 0x18
Trying address: 0x19
Trying address: 0x1A
Trying address: 0x1B
Trying address: 0x1C
Trying address: 0x1D
Trying address: 0x1E
Trying address: 0x1F
Trying address: 0x20
Trying address: 0x21
Trying address: 0x22
Trying address: 0x23
Trying address: 0x24
Trying address: 0x25
Trying address: 0x26
Trying address: 0x27
Trying address: 0x28
Trying address: 0x29
Trying address: 0x2A
Trying address: 0x2B
Trying address: 0x2C
Trying address: 0x2D
Trying address: 0x2E
Trying address: 0x2F
Trying address: 0x30
Trying address: 0x31
Trying address: 0x32
Trying address: 0x33
Trying address: 0x34
Trying address: 0x35
Trying address: 0x36
Trying address: 0x37
Trying address: 0x38
Trying address: 0x39
Trying address: 0x3A
Trying address: 0x3B
Trying address: 0x3C
Trying address: 0x3D
Trying address: 0x3E
Trying address: 0x3F
Trying address: 0x40
Trying address: 0x41
Trying address: 0x42
Trying address: 0x43
Trying address: 0x44
Trying address: 0x45
Trying address: 0x46
Trying address: 0x47
Trying address: 0x48
Trying address: 0x49
Trying address: 0x4A
Trying address: 0x4B
Trying address: 0x4C
Trying address: 0x4D
Trying address: 0x4E
Trying address: 0x4F
Trying address: 0x50
Trying address: 0x51
Trying address: 0x52
Trying address: 0x53
Trying address: 0x54
Trying address: 0x55
Trying address: 0x56
Trying address: 0x57
Trying address: 0x58
Trying address: 0x59
Trying address: 0x5A
Trying address: 0x5B
Trying address: 0x5C
Trying address: 0x5D
Trying address: 0x5E
Trying address: 0x5F
Trying address: 0x60
Trying address: 0x61
Trying address: 0x62
Trying address: 0x63
Trying address: 0x64
Trying address: 0x65
Trying address: 0x66
Trying address: 0x67
Trying address: 0x68
Trying address: 0x69
Trying address: 0x6A
Trying address: 0x6B
Trying address: 0x6C
Trying address: 0x6D
Trying address: 0x6E
Trying address: 0x6F
Trying address: 0x70
Trying address: 0x71
Trying address: 0x72
Trying address: 0x73
Trying address: 0x74
Trying address: 0x75
Trying address: 0x76
Trying address: 0x77
Trying address: 0x78
Trying address: 0x79
Trying address: 0x7A
Trying address: 0x7B
Trying address: 0x7C
Trying address: 0x7D
Trying address: 0x7E

BSP Library : 0.9.3
Bootloader  : s132 6.1.1 r1
Serial No   : F7BCC6FAA56CCB4E
```

closed 2019-04-04 by beegee-tokyo

ladyada, 2019-04-04:
just to clarify - its not really a hardware issue with feather, i2c is just not specified to work without pullups, any extra capacitance on the lines will affect it. don't run i2c without pullups because this can *always* happen! :)

EmilePapillon, 2024-12-09:
The feather bords come with pullups according to [this](https://learn.adafruit.com/bluefruit-nrf52-feather-learning-guide/device-pinout#i2c-pins-2860579)

dhalbert, 2024-12-09:
> The feather bords come with pullups according to [this](https://learn.adafruit.com/bluefruit-nrf52-feather-learning-guide/device-pinout#i2c-pins-2860579)

Feather microcontroller boards do _not_ come with pullups by default. If there are on-board devices sharing the external I2C bus, then there will be pullups, but in general, Adafruit boards expect the I2C breakout boards to come with appropriate pullups.

Quoting from the link above:
> I2C pins on the nRF52832 require external pullup resistors to function, which are not present on the Adafruit nRF52 Feather by default. You will need to supply external pullups to use these. All Adafruit I2C breakouts have appropriate pullups on them already, so this normally won't be an issue for you.