adafruit/Adafruit_nRF52_Arduino #243: Software serial errors when bluetooth is advertising

jps2000, 2019-03-21:
Software serial bit banging is corrupted when BLE is adverising
https://forums.adafruit.com/viewtopic.php?f=57&t=147186

nrf52832  feather  Software 0.9.03
bleuart example
software serial  @ 9600 baud 
GPS module is sending data

With calling Bluefruit.Advertising.start(0); I get corrupt data like

$GPRMC,001509.870,V,,,⸮⸮r⸮⸮b⸮r⸮⸮b⸮⸮⸮⸮bbbr⸮V*3A
$GPVTG,0.00,T,,M,0.00,N,0.00,K,N*32
$GPGGA,001509.870,,,,,0,0,,,M,,M,,*4A
$GPGSA,A,1,,,,,,,,,,,,,,,,1*03
$GPGSA,A,1,,⸮⸮bbbbbbbbbbb⸮R⸮⸮j
$GPGSV,1,1,00,0*65
$GLGSV,1,1,00,1*78
$GPGLL,,,,,001509.870,V,N*78

when  Bluefruit.Advertising.start(0); is commented out then data are ok:

$GPRMC,000036.868,V,,,,,0.00,0.00,060180,,,N,V*3B
$GPVTG,0.00,T,,M,0.00,N,0.00,K,N*32
$GPGGA,000036.868,,,,,0,0,,,M,,M,,*4B
$GPGSA,A,1,,,,,,,,,,,,,,,,1*03
$GPGSA,A,1,,,,,,,,,,,,,,,,2*00
$GPGSV,1,1,00,0*65
$GLGSV,1,1,00,1*78
$GPGLL,,,,,000036.868,V,N*79

Probably caused by freeRTOS
--> mutex guard may help?

ladyada, 2019-03-21:
this is going to be a very low priority to fix because softwareserial is a huge hack - you could try a GPS with I2C or an nrf52840 which has a hardware serial

jps2000, 2019-03-22:
I understand  what very low priority means. 
I suggest to mention  this  somewhere like here:
https://github.com/adafruit/Adafruit_nRF52_Arduino/tree/master/libraries/SoftwareSerial

It works in Arduino primo due to different OS and this I was not aware of.

Thank you

ladyada, 2019-03-22:
added
" Software Serial will not work when Bluetooth Radio is active because the softdevice
 interrupts the library."

to the examples and to the FAQ

closed 2019-03-22 by ladyada