adafruit/Adafruit_nRF52_Arduino #246: Redefine of NFC pins to GPIO causing difficulty on Feather BLE

Protonerd, 2019-04-03:
Hi guys,

I hope someone can help me out here.
I'm having a problem redefining the Feather BLE's NFC pins from NFC to GPIO. I edited the boards.txt under ..\Adafruit_nRF52_Arduino-master to add under the
#**********************************************
Feather nRF52832 Board
#**********************************************
...
Build
...
feather52832.build.extra_flags=-DNRF52832_XXAA -DNRF52 -DCONFIG_NFCT_PINS_AS_GPIOS
In addition I added a #define CONFIG_NFCT_PINS_AS_GPIOS in the application code as well.

But it still refuses to toggle the 2 NFC pins as GPIOs. What am I doing wrong?

ladyada, 2019-04-03:
hiya the boards already ship with NFC as GPIO?

dhalbert, 2019-04-03:
`CONFIG_NFCT_PINS_AS_GPIOS` must be defined in a place that `nrfx/mdk/system_nrf52.c` will see it, for this to work. That file includes very few other files, so a `#define` in your app code won't make any difference. So check the `gcc` command line for `system_nrf52.c` (assuming it is compiled).

hathach, 2019-04-04:
@Protonerd you should ask question on forum support, this is answered there before. You need to run this sketch once https://github.com/adafruit/Adafruit_nRF52_Arduino/tree/master/libraries/Bluefruit52Lib/examples/Hardware/nfc_to_gpio

closed 2019-04-04 by hathach

Protonerd, 2019-04-05:
Thanks guys, the above mentioned sketch solved the problem.