adafruit/Adafruit_nRF52_Arduino #250: Wrong values in g_ADigitalPinMap?

henrygab, 2019-04-11:
It appears that five pins are listed twice in the pin map for the Feather nRF52840  Express:
https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/b6a8566a531c11bc21b9404615b1a6c836e7ddef/variants/feather_nrf52840_express/variant.cpp#L34-L36
https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/b6a8566a531c11bc21b9404615b1a6c836e7ddef/variants/feather_nrf52840_express/variant.cpp#L38
https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/b6a8566a531c11bc21b9404615b1a6c836e7ddef/variants/feather_nrf52840_express/variant.cpp#L44

These pins are listed a second time at their own index:
https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/b6a8566a531c11bc21b9404615b1a6c836e7ddef/variants/feather_nrf52840_express/variant.cpp#L70-71

Is this intentional?  This seems like it may be a bug....

henrygab, 2019-04-12:
In addition to the double-listing for those five pins, the following five pins are not listed.  Here is the list of missing ports:
* 0.00 -- GPIO (might be external 32.768kHz crystal (XL1)?) 
* 0.01 -- GPIO (might be external 32.768kHz crystal (XL2)?)
* 0.06 -- GPIO (high-frequency enabled, high-drive enabled)
* 0.09 -- GPIO (standard drive, low-frequency only … <= 10kHz)
* 0.18 -- GPIO (or QSPI/CSN, or configured as system reset)

Of course, the feather express board might be using one or more of those port.pin for crystal, reset button … but currently they are not exposed as a digital pin for Arduino.

hathach, 2019-04-16:
thanks @henrygab it is indeed an issue, the header declare more pins that the board actually expose. I will check out the schematics with your PR.

commit e343c8a5 (2019-04-19): Merge pull request #251 from henrygab/patch1

Fix #250 -- wrong values for g_ADigitalPinMap
commit ee9cbc84 (2019-04-19): Fix #250 -- wrong values for g_ADigitalPinMap
closed 2019-04-19 by hathach