adafruit/Adafruit_nRF52_Arduino #252: Serial1 crashes NRF52840 dongle (PCA10059)

RoganDawes, 2019-04-18:
- Please provide details that help us to reproduce your issue

I am trying to use the built-in UART on the NRF52840 dongle (PCA10059), and tested it with the example code here: https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/libraries/Bluefruit52Lib/examples/Hardware/Serial1_test/Serial1_test.ino

Since there does not seem to be a board entry for the PCA10059 in v0.10.1 of the NRF52 support pack, I have chosen ```Bluefruit nRF52840```. I guess this is where my problem lies!

However, the instant I connect to the USB Serial port, the USB device disconnects from the host, and is subsequently unresponsive to even the reset button. I assume this is because the pins used for the RX or TX on the default 52840 board are connected to something else on the PCA10059?

  - Your Board and BSP version
    Nordic NRF52840 dongle - PCA10059

  - Your OS (mac/windows/linux) and its version
Linux (ubuntu 16.04). Arduino 1.8.9, Adafruit NRF52 by Adafruit version 0.10.1.

  - Steps to reproduce, or better your (simplified) sketch if possible
I am using the provided example: https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/libraries/Bluefruit52Lib/examples/Hardware/Serial1_test/Serial1_test.ino

  - Serial debug log (enable IDE's Debug Mode Level to 1 or 2)  

There is nothing in the debug log. The instant ```while (!Serial) delay(10);``` returns true, the USB device disconnects.

From ```dmesg```:

```
[366744.184134] usb 1-5: new full-speed USB device number 10 using xhci_hcd
[366744.334628] usb 1-5: New USB device found, idVendor=239a, idProduct=8029
[366744.334634] usb 1-5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[366744.334637] usb 1-5: Product: Bluefruit nRF52840
[366744.334641] usb 1-5: Manufacturer: Adafruit Industries
[366744.334644] usb 1-5: SerialNumber: 8A7410F6A04BE65B
[366744.335859] cdc_acm 1-5:1.0: ttyACM0: USB ACM device
[366755.124466] usb 1-5: USB disconnect, device number 10
[366755.124667] cdc_acm 1-5:1.0: failed to set dtr/rts
```

hathach, 2019-04-18:
pca10059 is not supported since I dont have hw ready to test with. pca10056 is also not officially supported. Maybe others with the same hw could help.

RoganDawes, 2019-04-18:
I actually figured out a workaround. As mentioned, the problem was the default pin definition. Some poking around revealed the "Uart.h" source code, with a ```setPins(rx, tx)``` method.

Through an iterative process, I figured out that using ```Serial1.setPins(xx, 42);``` would allow me to transmit the debugging output that I wanted from P1_10 to another USB-UART. However, trying the same iterative technique with ```yes >> /dev/ttyUSB3``` and stepping through the possible pin numbers did not result in any input on that port. Not quite sure what I was doing wrong!

Anyway, I guess this can be closed now.

closed 2019-04-18 by RoganDawes