adafruit/Adafruit_nRF52_Arduino #255: Suspicious line continuation in comment in BLEHidGeneric.cpp

MacGyverNL, 2019-04-27:
I'm not sure whether this is actually an issue. I was reading the source code of BLEHidGeneric.cpp to figure out the correct keycodes to use for some special keys and I came across this line continuation that looks suspicious: https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/68400a76662af268829e3c6c66ae62ac02eaae76/libraries/Bluefruit52Lib/src/services/BLEHidGeneric.cpp#L520

The effect is that the next line is also a comment. Now I'm not versed well enough in keycodes to know whether that's intentional, but it's highly unusual as commenting styles go so if somebody could confirm this is or is not a bug...

bjc, 2019-04-27:
It's not a bug. That's not a line-comment (`// ...`) but an inline one (`/* ... */`). It's weird, but not broken.

MacGyverNL, 2019-04-27:
Notice the `//` in front of the `/* ... */` block and the `\` at the end of the line. `\` at the end of the line is what is known as a line continuation, and in Arduino (and vanilla C++), a line continuation at the end of a line-comment started with `//` *continues the comment on the next line*. The `/* ... */` is not relevant to the (potential) bug.

Minimal testcase of this, just paste it literally in the Arduino IDE:
```
// Just to prove that a line continuation at the end of a comment makes the next line a comment \
void setup() {
}
```
This will not compile, with the tokenizer complaining `expected declaration before '}' token`.

So the question is, is line 521 *supposed to be* a comment in its entirety?

hathach, 2019-04-27:
Thank you very much for spotting this, you got some eagle eye there. It is left over when I move this table around and apply text replacement. Fixed already :).

closed 2019-04-27 by hathach