adafruit/Adafruit_nRF52_Arduino #266: Can't burn bootloader on macOS via ArduinoIDE

NaotoFujihiro, 2019-05-09:
- ## Issues
I tried to burn bootloader based on the following steps.
https://github.com/adafruit/Adafruit_nRF52_Arduino#burning-new-bootloader

However the following error occured and I couldn't burn it.

```
Arduino: 1.8.9 (Mac OS X), Board: "Adafruit Bluefruit nRF52840 Feather Express, 0.2.9 (s140 6.1.1), Level 0 (Release)"

nrfjprog --program ~/Library/Arduino15/packages/adafruit/hardware/nrf52/0.10.1/bootloader/feather_nrf52840_express/feather_nrf52840_express_bootloader-0.2.9_s140_6.1.1.hex -f nrf52 --chiperase --reset 
java.io.IOException: Cannot run program "nrfjprog": error=2, No such file or directory
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at java.lang.Runtime.exec(Runtime.java:620)
	at java.lang.Runtime.exec(Runtime.java:485)
	at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:11)
	at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:129)
	at cc.arduino.packages.uploaders.SerialUploader.runCommand(SerialUploader.java:383)
	at cc.arduino.packages.uploaders.SerialUploader.burnBootloader(SerialUploader.java:375)
	at processing.app.Editor.lambda$handleBurnBootloader$65(Editor.java:2384)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.io.IOException: error=2, No such file or directory
	at java.lang.UNIXProcess.forkAndExec(Native Method)
	at java.lang.UNIXProcess.<init>(UNIXProcess.java:247)
	at java.lang.ProcessImpl.start(ProcessImpl.java:134)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 8 more
Error while burning bootloader.
```

I've read the same issue #149.

- ## How to fix it
1. Burn bootloader via terminal
If I burn bootloader using terminal, the procedure ends successfully.
```
$ nrfjprog --program ~/Library/Arduino15/packages/adafruit/hardware/nrf52/0.10.1/bootloader/feather_nrf52840_express/feather_nrf52840_express_bootloader-0.2.9_s140_6.1.1.hex -f nrf52 --chiperase --reset 
```

2. Edit programmers.txt
To burn bootloader via ArduinoIDE, I edited `programmers.txt` in the packages installed from Boards Manager.

```
nrfjprog.program.cmd=nrfjprog
nrfjprog.program.cmd.macosx=/usr/local/bin/nrfjprog
nrfjprog.program.cmd.windows=nrfjprog.exe
```

```
/usr/local/bin/nrfjprog --program ~/Library/Arduino15/packages/adafruit/hardware/nrf52/0.10.1/bootloader/feather_nrf52840_express/feather_nrf52840_express_bootloader-0.2.9_s140_6.1.1.hex -f nrf52 --chiperase --reset 
Parsing hex file.
Erasing user available code and UICR flash areas.
Applying system reset.
Checking that the area to write is not protected.
Programming device.
Applying system reset.
Run.
```

**Editing `programmers.txt` in Adafruit_nRF52_Arduino repository ended in  vain.**

- ## Environment
```
$ nrfjprog -v
nrfjprog version: 9.8.1
JLinkARM.dll version: 6.44h
```

ArduinoIDE: 1.8.9
Board: Adafruit Bluefruit nRF52840 Feather Express
Programmer: J-Link for Bluefruit nRF52
Boards Manager: Adafruit nRF52 by Adafruit version 0.10.1
Hardware: nRF52840 Dongle

```
$ sw_vers
ProductName:	Mac OS X
ProductVersion:	10.14.4
BuildVersion:	18E226
```

Best regards.
Thanks.

hathach, 2019-05-09:
This is normal, on macos Arduino IDE has its own limited PATH compare to user PATH at terminal. You have to make a symlink, it is mentioned in the arduino repo
https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/README.md

closed 2019-05-09 by hathach

NaotoFujihiro, 2019-05-09:
Yes I know it.
And I made a symlink, therefore I could run `nrfjprog -v`.

Thanks anyway.

hathach, 2019-05-09:
Np, this issue can ended up helping other with similar issue 👍👍