adafruit/Adafruit_nRF52_Arduino #269: nRF52832 hid_keyboard is not work

shaoyuan1943, 2019-05-14:
I am beginer for nRF52832 dev. I want to build a bluetooth keyboard by nRF52832. I use this code: https://github.com/adafruit/Adafruit_nRF52_Arduino/tree/master/libraries/Bluefruit52Lib/examples/Peripheral/hid_keyboard

when nRF52832 is running, my desktop(windows 10) found bluetooth and display bluetooth is paired and connected, but actually in nRF52832 board is not connected, I invoke this code: Bluefruit.Central.connected(); it return false, and connected callback is not invoked, so the HID is not working. The following code is added by myself:

void setup()
{
  Serial.begin(115200);
  while ( !Serial ) delay(10);   // for nrf52840 with native usb

  Serial.println("Bluefruit52 HID Keyboard Example");
  Serial.println("--------------------------------\n");
 

  Bluefruit.begin();
  Bluefruit.setTxPower(4);    // Check bluefruit.h for supported values
  Bluefruit.setName("Bluefruit52");

  // Configure and Start Device Information Service
  bledis.setManufacturer("Adafruit Industries");
  bledis.setModel("Bluefruit Feather 52");
  bledis.begin();
 
  // following two lines code added by myself	
  Bluefruit.Central.setConnectCallback(cent_connect_callback);
  Bluefruit.Central.setDisconnectCallback(cent_disconnect_callback);

  /* Start BLE HID
   * Note: Apple requires BLE device must have min connection interval >= 20m
   * ( The smaller the connection interval the faster we could send data).
   * However for HID and MIDI device, Apple could accept min connection interval
   * up to 11.25 ms. Therefore BLEHidAdafruit::begin() will try to set the min and max
   * connection interval to 11.25  ms and 15 ms respectively for best performance.
   */
  blehid.begin();

  // Set callback for set LED from central
  blehid.setKeyboardLedCallback(set_keyboard_led);

  /* Set connection interval (min, max) to your perferred value.
   * Note: It is already set by BLEHidAdafruit::begin() to 11.25ms - 15ms
   * min = 9*1.25=11.25 ms, max = 12*1.25= 15 ms
   */
  /* Bluefruit.Periph.setConnInterval(9, 12); */

  // Set up and start advertising
  startAdv();
}

// following code added by myself
void cent_connect_callback(uint16_t conn_handle)
{
  Serial.println("[Cent] Connected");
}

// following code added by myself
void cent_disconnect_callback(uint16_t conn_handle, uint8_t reason)
{
  Serial.println("[Cent] Disconnected");
}

SOME INFO: On windows 10, [Control Panel]->[Hardware And Sound]->[Device And Printer], I found Bluefruit52 ble device, I saw this device property is [not enabled], the [Service] is empty.

hathach, 2019-05-17:
can you fill out the information in the bug report template

**Set up (please complete the following information)**
  - Your Board and BSP version
  - Your OS (mac/windows/linux) and its version
  - Serial debug log (enable IDE's Debug Mode Level to 1 or 2)

shaoyuan1943, 2019-05-17:
sorry, I'm made mistake for that, it's not bug, is't my fault.

closed 2019-05-17 by shaoyuan1943