adafruit/Adafruit_nRF52_Arduino #272: nRF52840 std::map.insert()

czuvich, 2019-05-22:
I'm trying the std library on the nRF52840 Express.  I am able to instantiate the std::vector and std::map just fine.  I can also invoke the push_back() method of std::vector; however, when I try to invoke the std::map.insert() method I get the following exception.

`var/folders/_d/ksp5fdyx5h1flr8s6l6rvgj00000gn/T/arduino_build_547782/sketch/FreeRTOSTimer.cpp.o: In function `std::_Rb_tree_iterator<std::pair<void* const, FreeRTOSTimer*> >::operator--()':
/Users/Clay/Library/Arduino15/packages/adafruit/tools/gcc-arm-none-eabi/5_2-2015q4/arm-none-eabi/include/c++/5.2.1/bits/stl_tree.h:220: undefined reference to `std::_Rb_tree_decrement(std::_Rb_tree_node_base*)'`

I think it has something to do with the compiler flags.  Is there any way I can enable exceptions and std::map.insert() on the nRF52840 Express?

czuvich, 2019-05-23:
Adding the following lines fixed my issues.

-lsupc++_s -lm -lc_s -lstdc++_s

czuvich, 2019-05-23:
Ok.. so I was able to get the linker error to go away with the following line

compiler.c.elf.cmd=arm-none-eabi-g++

I replaced the gcc with g++.  Any reasons why you do not want g++?

hathach, 2019-07-09:
thanks for your effort, though we don't plan to provide support for STL library. We would be happy to review and merge if you could submit a PR.

czuvich, 2019-07-12:
Ok.  It'll be my first PR :)  Thanks for the update!

closed 2019-07-12 by czuvich

bojh, 2019-07-12:
I would appreciate that. Some weeks ago I tried to run the [Eigen ](https://github.com/bolderflight/Eigen.git)* for Arduino library in the adaptation of bolderflight. While under ESP32, it directly compiles, in the nRF52 core the #defines min(a,b), max(a,b) in core/nRF5/Arduino.h has to be uncommented for a ugly fix. Does somebody know a better context sensitive solution?
*) algebra library i.e. base for matrix manipulation/handling. (...for Kalman filter). Starting from nRF52840 also an adaptation of [MiniDNN ](https://github.com/yixuan/MiniDNN.git)would be nice (but it has currently additional compile problems).

czuvich, 2019-07-12:
In regards to the min/max issue, I just ended up creating a "Defines.h" that undefines the min/max functions in Arduino.h.  I then reference the Defines.h everywhere I need Arduino.h.  It's hacky I know.  Probably no help to you, but just thought I'd post.

```
#include "Arduino.h"
#undef min
#undef max
```
In regards to the std library, you can change ```compiler.S.cmd=arm-none-eabi-gcc``` to ```compiler.S.cmd=arm-none-eabi-g++```.  You'll have to hack in a few other things in your code depending on usage as well.  You're not going to get the full std library, but you'll get a good bit.

bojh, 2019-07-12:
good workaround to keep Arduino.h untouched -THX

hathach, 2019-07-12:
would an wrap around `max,min` with `#ifndef max, #ifndef min` fixes the issue with Eigen ? If possible @czuvich @bojh either of you could submit an PR to fix that (please provide more information of the conflict as well). We will review and merge.