adafruit/Adafruit_nRF52_Arduino #273: [Feature] analogRead(VDD)

ericlangel, 2019-05-23:
The NRF52832 has an internal ADC Channel for VDD

i tried to change the "wiring_analog_nRF52.c" file

`uint32_t analogRead( uint32_t ulPin )
{
  uint32_t pin = SAADC_CH_PSELP_PSELP_NC;
  uint32_t saadcResolution;
  uint32_t resolution;
  int16_t value;

  if (ulPin >= PINS_COUNT) {
    return 0;
  }

  ulPin = g_ADigitalPinMap[ulPin];

  switch ( ulPin ) {
    case 2:
      pin = SAADC_CH_PSELP_PSELP_VDD;
      break;

    case 3:
      pin = SAADC_CH_PSELP_PSELP_AnalogInput1;
      break;

    case 4:
      pin = SAADC_CH_PSELP_PSELP_AnalogInput2;
      break;

    case 5:
      pin = SAADC_CH_PSELP_PSELP_AnalogInput3;
      break;

    case 28:
      pin = SAADC_CH_PSELP_PSELP_AnalogInput4;
      break;

    case 29:
      pin = SAADC_CH_PSELP_PSELP_AnalogInput5;
      break;

    case 30:
      pin = SAADC_CH_PSELP_PSELP_AnalogInput6;
      break;

    case 31:
      pin = SAADC_CH_PSELP_PSELP_AnalogInput7;
      break;

    default:
      return 0;
  }`

When i'am do analogRead(2), it samples the internal VDD Voltage.
It's just a quick&dirty Solution, but maybe you can add an 9th ADC Channel for VDD with a good API

pyro9, 2019-07-29:
Just a thought, perhaps choose a large virtual pin number so that no real device will ever need it (e.g. 0xfffffffe) and assign it as the internal ADC channel. Then add it to g_ADigitalPinMap in each varient and in the case statement for analogRead?

jpconstantineau, 2020-04-23:
Someone posted this gist: https://gist.github.com/vladkozlov69/2500a27cd93245d71573164cda789539

I tried it and it works great.

tusharvb19, 2020-08-28:
How to measure VDD by api call like analogRead(VDD)?

ericlangel, 2020-09-01:
change code in wiring_analog_nRF52.c

tusharvb19, 2020-09-06:
Can someone help me and share the snippet to measure the internal VDD?

ericlangel, 2020-09-07:
hello tusharvb19,

what exactly is you problem?

code snippet is shared in the first posting of this issue.

Did you change "wiring_analog_nRF52.c"?
is the code compiling?
which Version u are using?

hathach, 2020-10-12:
fixed by #591

closed 2020-10-12 by hathach