adafruit/Adafruit_nRF52_Arduino #275: How can I make my nRf52832 server support multiple central connection?

TSOgang, 2019-05-28:
Im pretty new to BLE and I've been using the Bluefruit libraries and the Adafruit52832 feather board for my first programs. So far I can use my phone as a client to connect to it and write/read characteristics. I read that the nRF52832 can support up to around 20 connections as both peripheral and central. Now, that's fine, but so far I haven't found anything in the Bluefruit libraries to help me set up multiple central connections to my peripheral and vice-versa. So my question is; how can I set-up multiple connections to a single nRF52832  (peripheral role). 

It is entirely possible that I am blind and missed oblivious information.
Thank you in advance!

tilink, 2019-06-22:
Hey @TSOgang - I think the "central_bluart_multi" or the "bluart_multi" examples might be helpful for you? I'm trying it right now, no luck so far integrating multiple peripherals into my project that I built up for one peripheral but we'll see if I can figure it out.

hathach, 2019-06-25:
there is example for multiple central link, for more question on how it works, please post question to our support forum
https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/libraries/Bluefruit52Lib/examples/Central/central_bleuart_multi/central_bleuart_multi.ino

closed 2019-06-25 by hathach