adafruit/Adafruit_nRF52_Arduino #276: [Bug] BluefruitFS and SD.h

czuvich, 2019-05-29:
BSP Library : 0.10.1
Board: nRF52840 Express
Bootloader  : s140 6.1.1

I am using an SD Card and the InternalFS on the nRF52840.  I was trying out the Internal_ReadWrite example for the nRF52840.  The example as-is runs just fine; however, I added an ```#include <SD.h>``` and prefixed the file local variable with the namespace ```BluefruitFS::File file(InternalFS);```.

Instead of writing out the text of ```#define CONTENTS```, nothing is output from reading the file.  I think there are some conflicts in the internal library and using ```SD.h```?

czuvich, 2019-05-29:
I also wanted to add, so far I don't think I can use an external SD Card nor can I use the ExternalFS with the nRF52840.  I'm kind of stuck using the InternalFS for any data persistence.

hathach, 2019-05-30:
thanks, we are reworking the filesystem. This should be resolved later on.

commit b99cae32 (2019-05-30): fix #276 rename macro FILE_READ/WRITE to enum FILE_O_READ/WRITE

to avoid conflict with other filesystem define
closed 2019-05-31 by hathach