adafruit/Adafruit_nRF52_Arduino #279: BLEClientHidAdafruit compile error

seihan, 2019-05-31:
'HID_KEYCODE_TO_ASCII' was not declared in this scope

board: nRF52832, nRF52840
using Arduino 1.8.8

hathach, 2019-05-31:
can you also tell us which bsp version and which example you got issue with

seihan, 2019-05-31:
Yes, of course. I'm sorry.

BSP origin / master, central_hid example

`Arduino/hardware/Adafruit/Adafruit_nRF52_Arduino/libraries/Bluefruit52Lib/examples/Central/central_hid/central_hid.ino: In function 'void processKeyboardReport(hid_keyboard_report_t*)':
central_hid:209:24: error: 'HID_KEYCODE_TO_ASCII' was not declared in this scope
         ch = shifted ? HID_KEYCODE_TO_ASCII[kc].shifted : HID_KEYCODE_TO_ASCII[kc].ascii;`

hathach, 2019-05-31:
You should only report the bug with release, master or develop is still work in progress. It is the conflict between ble and usb hid, I am focusing on usb just now, But thanks for the issue.

commit 79252a5c (2019-06-06): Merge pull request #283 from adafruit/develop

fix #279 conflict between ble hid and usb hid
closed 2019-06-06 by hathach