adafruit/Adafruit_nRF52_Arduino #281: [Bug] BLE Central Crash on Bluefruit.disconnect()

czuvich, 2019-06-05:
BSP Library : 0.10.1
Board: nRF52840 Express
Bootloader : s140 6.1.1

I'm seeing a crash on my nRF Central when I try to run a few reads and a few writes consecutively against another nRF Peripheral (both nRF52840 Express chips).

I'm not sure if this issue is related to the Bluefruit framework; however, I think it's related to congestion on the BLE stack.  I do the following:

1) Scan
2) Connect to the Peripheral
3) Read 14 bytes from service 1, characteristic 1
4) Read 4 bytes from service 1, characteristic 2
5) Read 8 bytes from service 2 characteristic 3
6) Write 4 bytes from service 3, characteristic 4
7) Write 1 byte from service 4, characteristic 5
8) Disconnect the central via Bluefruit.disconnect(handle)

Everything seems to be ok up until the disconnect().  Once the disconnect occurs, the nRF central crashes.  I added a small delay between each read/write (7ms), and it seems to have cleared things up.  I've seen congestion issues before on the ESP32; however, I wasn't sure if it'd be a problem on the nRF52840.  

Perhaps it's something I'm doing wrong?  Or is that just the nature of BLE?  I also saw the same issue if the Peripheral issued a disconnect to the client (same API).  Adding some delays seems to fix the problem; however, I hate magic numbers like that.

czuvich, 2019-06-05:
Further investigation shows that there is definitely something specific with disconnection from a Peripheral/Central.  Is there a recommended approach for disconnecting a Central from the Peripheral?  I'm going to try adding a semaphore that waits on the Disconnect() event.

hathach, 2019-06-05:
Could you also post which sketch you are using with both central and prph side, it is best if they are from the stock examples.

czuvich, 2019-06-06:
I'll go ahead and close this issue.  I created a small sketch for central and perph and it appears to be working fine.  It looks like I just need to process my scans differently.  I didn't realize the pointers (the scan reports) were being invalidated on the next execution cycle.  I'm used to the ESP32 processing which is slightly different (it feels like Nordic "expects" you to process scans as they come in rather than building them up and post-processing).  Thanks.

closed 2019-06-06 by czuvich

czuvich, 2019-06-06:
Ok I figured it out.  I had a couple of issues.

1) Do not wait on a semaphore on any callbacks from Bluefruit. It will hose the ada_callback loop (I was doing this in the scanner callback to block while I connect to a device.. which was blocking the connect callback).
2) Passing a timeout into the Scanner is a little misleading. For instance:

1) Scanner starts scanning and is expected to timeout in 5s with a autoRestart set to true.
2) A device is found in the scan callback.
3) Connect to the device, do some stuff, disconnect.
4) The scanner restarts; however, it restarts based on the original timeout.  

This behavior is what was causing some issues in my app.  I was expecting the Scanner to timeout after 5s; however, if you continue to connect/disconnect devices the scanner may never actually timeout. 

The way I am going to remedy that is to add my own timer that puts a stop on scanning via RTOS.

hathach, 2019-06-07:
Thanks for updating your info on situation. For a callback, it is normally called within ada callback task, or can opt to be call within ble task. This give some option for synchronization. For more complicated sketch, you should wait for semaphore/queue in loop() and post/send event in the callback.

czuvich, 2019-06-07:
Thanks @hathach.  One last note... I was also hitting a crash in my app since I was using the BLEClientCharacteristics and ClientServices in a method.  I think what was happening was that the bluefruit framework is cleaning up connection handles on the DISCONNECT call.  It's doing that by referencing "this" for the BLEClientServices/Char.  Well, since I kept the variables in method scope it was causing an issue (I think that was the problem).

I resolved this other issue by putting the ClientChars/Services into a static container and calling begin() early in the app lifecycle.  I'm not entirely sure if this will cause issues connecting to different peripherals, but we'll see.

hathach, 2019-06-07:
Glad you figure it out, writing code to sync  multi threads with rtos can be challenging. When in doubt, dive into the code behind the scene and don't hesitate to post your question/thought on our support forum. People will come and help