adafruit/Adafruit_nRF52_Arduino #285: bleuart example not working w iOS app

CollinCunningham, 2019-06-15:
**Describe the bug**
When running the bleuart.ino example sketch, UART appears busy/unavailable to the Bluefruit LE Connect app. Commenting out [line 55](https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/e690442c7fa1e8b0249230800b5ca268058cfd34/libraries/Bluefruit52Lib/examples/Peripheral/bleuart/bleuart.ino#L55) fixes the issue.

**Set up (please complete the following information)**
  - Feather nRF52832, BSP v0.11.0
  - macOS 10.14.5, Bluefruit LE Connect v3.4.0, iOS 12.3.1

**To Reproduce**
1. Run examples/Peripheral/bleuart/bleuart.ino
2. Connect via Bluefruit LE Connect
3. Open UART module
4. Bottom text field is disabled & data sent from Arduino serial monitor is not displayed

**Expected behavior**
Ability to send receive data over UART

**Additional context**
Let me know if this looks like an issue on the iOS side - we can roll a fix into the next update.

hathach, 2019-06-17:
@CollinCunningham hmm, this is strange, I couldn't reproduce this with my iphone 6 on iOS 11.3 with Bluefruit 3.4.0.  I am upgrading it to ios 12.3.1 and try again.

hathach, 2019-06-17:
@CollinCunningham do you have other iphone to test with, just make sure it is not something specific with your iphone.

CollinCunningham, 2019-06-17:
@hathach weird - I'll locate another iphone to test. Thanks for checking.

dhalbert, 2019-06-17:
Try iPhone 6 or 6s vs 7 or later. 6 and 6s don't have extended-length packet support in hardware.

dhalbert, 2019-06-17:
I noticed this on my iPhone 6S vs a newer iPad, I think.

CollinCunningham, 2019-06-17:
I'm on an iphone 7+, fwiw

dhalbert, 2019-06-17:
See https://github.com/adafruit/circuitpython/pull/1491 for a fix I had to put into CircuitPython. We weren't handling `BLE_GAP_EVT_DATA_LENGTH_UPDATE_REQUEST`. Also was failing to handle `BLE_GATTS_EVT_SYS_ATTR_MISSING`, sent by Android devices.

hathach, 2019-06-18:
@CollinCunningham please also enable Debug Level to 2 to see if there is any abnormal activity. Thanks @dhalbert my wife's phone is iphoneX, we also already handle both events you mentioned (used to got issue with LUN UPDATE REQEUST as well)
https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/libraries/Bluefruit52Lib/src/bluefruit.cpp#L865

CollinCunningham, 2019-06-19:
@hathach just getting back to this - tested and working on an iPhone X I tried yesterday. 
How do I enable debug level 2? I tried adding `#define CFG_DEBUG 2` to the sketch, but I'm not seeing any logs/etc.

CollinCunningham, 2019-06-19:
@hathach Ah - nevermind. Just found it in the IDE.

hathach, 2019-06-20:
@CollinCunningham your iPhone 7+ is bonded with the device, probably running blehid/midi/ancs previously. This can prevent iPhone to re-discover service when switching to normal sketch. Can you go into iOS bluetooth setting and **Forget** nrf52 device then refresh/off-then-on bluetooth and try again.

CollinCunningham, 2019-06-20:
@hathach thanks much - that was it 🎉
Strange that I've never encountered this before. I'll update the learn guide accordingly.

closed 2019-06-20 by CollinCunningham

hathach, 2019-06-20:
@CollinCunningham I am glad that help. The iOS behavior changes as well, it  is very tricky :)

CollinCunningham, 2019-06-20:
@hathach ya - there's been more than a few updates since I wrote iOS BLE code ;)