adafruit/Adafruit_nRF52_Arduino #288: Feather nRF52840 Express cannot connect to more than 4 peripheral BLE devices [Bug]

andy001sun, 2019-06-20:
**Describe the bug**
I tried to connect the Feather nRF52840 Express (central) to multiple peripheral BLE devices. However, the maximum peripheral number is 4, not the "BLE_MAX_CONNECTION 20" as in the datasheet and this library "bluefruit.h"

When I use Bluefruit.begin(0, 4) or less, it works well and can connect to 4 peripheral BLEs. 
When I set Bluefruit.begin(0, 5) or more, the BLE function cannot be enabled and the board works just like a common MCU without BLE function.

I tried both my own code and the example "central_bleuart_multi.ino", which are the same and the error is like the screenshot below:
[image]

**Set up**
  - Feather nRF52840 Express, BSP v0.11.0
  - Win 7 & 10

**To Reproduce**
Steps to reproduce the behavior:
1. Go to example 'central_bleuart_multi'
2. change 'Bluefruit.begin(0, 4)' to 5 or more
3. choose debug: "Level 2 (full debug)"
4. See error

**Expected behavior**
I want to use the central nRF52840 to connect to more peripherals (at least 5). Is there any way?

I checked the linker file (I'm not sure) "nRF52840_s140_v6.ld" but had no idea what to do. The file is below:
[image]

@hathach Could you please help? Thank you.

hathach, 2019-06-21:
It is not a bug, you need to provide more SRAM for SoftDevice for higher number of connections. The only current way is modify the linker. This could give you some hint https://learn.adafruit.com/bluefruit-nrf52-feather-learning-guide/hathach-memory-map#sram-layout-39-8

Check out Nordic docs for more details/info.

andy001sun, 2019-06-21:
@hathach Thank you! I will try to modify it.

nibraaska, 2023-07-09:
Were you able to modify it?