adafruit/Adafruit_nRF52_Arduino #289: [Bug] Using Bluetooth HID and USB HID at the same time is not possible

coyt, 2019-06-22:
I've built a Bluetooth keyboard with nRF52840 and it's working great using this framework. All hardware works - it's based on a feather nrf52840. USB CDC / the comm port is also working fine so usb hardware is definitely ok.   

Now, I'm trying to implement automatic switching between Bluetooth HID and USB HID. I want the keyboard to use USB HID when it's charging over a USB cable but switch to Bluetooth HID when cable is unplugged. 

I've dove into the included tinyUSB stack to try to implement this myself. It looks like this should be very easy:

1. Change tusb_config.h to turn CFG_TUD_HID & CFG_TUD_HID_KEYBOARD to on.
2. Use the functions in hid_device.h to make my keyboard - seems this library has very easy to use keyboard functions already
2a. tud_hid_keyboard_keycode(modifier, keycode);
2b. tud_hid_keyboard_key_release();

Unfortunately, I can't get this code to compile. There are tons of definition redeclaration errors. Diving further brings us to the bug:

It looks like the Adafruit bluetooth library is using a slightly modified COPY of the tinyUSB header file which defines all the HID descriptor stuff. Because it's a direct COPY, trying to include the USB HID functionality is not possible at the same time as the Bluetooth HID functionality because these header files try to re-declare each other and compiler complains. 

The two files:

1. .platformio\packages\framework-arduinoadafruitnrf52\libraries\Bluefruit52Lib\src\services\BLEHidGeneric.h

2. .platformio\packages\framework-arduinoadafruitnrf52\cores\nRF5\usb\tinyusb\src\class\hid\hid.h

Any chance this could be fixed? I'm not sure the best way to do it. Maybe implementing an interface layer to tinyUSB for HID as it appears you did with CDC "Adafruit_USBD_CDC.h"?

Thanks!

coyt, 2019-06-22:
Oh damn... this might not be an issue here after all...

It looks like you guys already fixed this. I'm using PlatformIO...so not exactly sure what version of adafruit bsp i have currently... but in my current version my BLEHidGeneric.h has all the definitions listed, but in the most recent version here this has been fixed.

Looks like it was fixed on Apr. 28th in cc430978e09d60a13c5c084235e93d22f53adf4a

I'll check with PlatofrmIO and see if they can update their copy and see if it fixes the problem.

hathach, 2019-06-25:
yeah, it is fixed previously in the latest release

closed 2019-06-25 by hathach