adafruit/Adafruit_nRF52_Arduino #290: [Bug/Question] Determine source of incoming serial info in bleuart_multi example

nicholasareed, 2019-06-22:
In the example [bleuart_multi example](https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/libraries/Bluefruit52Lib/examples/Peripheral/bleuart_multi/bleuart_multi.ino#L103) the loop for reading the incoming serial info doesn't show how to determine which connection the input comes from. Is it possible to know which central is sending the data? 

```
void loop()
{
  uint8_t buf[64];
  int count;

  // Forward data from HW Serial to BLEUART
  while (Serial.available())
  {
    // Delay to wait for enough input
    delay(2);
    count = Serial.readBytes(buf, sizeof(buf));

    printAll(buf, count);
  }

  // Forward from BLEUART to HW Serial
  while ( bleuart.available() )
  {
    count = bleuart.read(buf, sizeof(buf));
  
    printAll(buf, count);
  }
}
```

nicholasareed, 2019-06-24:
I found a solution using a callback (it also wasn't obvious to me that the `conn_handle` stays the same until disconnect):

in setup:
```
  // Configure and Start BLE Uart Service
  bleuart.begin();
  bleuart.setRxCallback(prph_bleuart_rx_callback);
```

and a new function: 

```
void prph_bleuart_rx_callback(uint16_t conn_handle)
{
  (void) conn_handle;

  Serial.print("--prph callback-- ");
  Serial.println(conn_handle);
  
  String fullstring = "";
  while ( bleuart.available() )
  {
    uint8_t ch;
    ch = (uint8_t) bleuart.read();
    fullstring += (char)ch;
  }
  
  Serial.print("[Prph] RX: ");
  Serial.println(fullstring);

}
```

closed 2019-06-24 by nicholasareed

hathach, 2019-06-26:
I am glad that you work it out yourself.