adafruit/Adafruit_nRF52_Arduino #293: [Bug] Central Scan Stop Callback does not fire.

Splitframe, 2019-06-27:
**Set up**^
Adafruit Feather NRF52832
BSP Library : 0.10.1
Bootloader  : s132 6.1.1
Serial No   : 8E9AB597AC611740 

**Expected behavior**
When I register a call back with:
`Bluefruit.Scanner.setStopCallback(centStopScanCallback);`
and then call
`Bluefruit.Scanner.stop();`
it should fire the callback for
`void centStopScanCallback()`

**Additional Information**
The only log prints aside from the header are 
`[BLE   ] BLE_GAP_EVT_ADV_REPORT : Conn Handle = 65535`

**Describe the bug**
I start the scan, but I don't want to connect inside the Scanner's RxCallback.
So I stop the scan after some time and decide to whom to connect in the stop call back.
The stop call back is never called though. I have a log print at the very start.

hathach, 2019-07-05:
@Splitframe thanks for file this issue, can you also upload the test sketch so that we could reproduce and fix thist quickly.

hathach, 2019-07-09:
btw, actively calling `Bluefruit.Scanner.stop()` won't trigger the stop callback. Stop callback will only be invoked when scanner is timed out specified in `Bluefruit.Scanner.start(timeout)`

closed 2019-07-09 by hathach

Splitframe, 2019-07-10:
Timeout didn't work for me at first either, but it seems I had a corrupted library. After I redownloaded the board libraries it worked though.

Anyway, shouldn't the callback be named something around the lines of "ScanTimeoutCallback" then?
Thanks for the help though.

hathach, 2019-07-10:
TimeoutCallback could be a better name, though stopCallback isn't too bad :)