adafruit/Adafruit_nRF52_Arduino #294: [Bug] Unable to init Bluefruit - nRF52840 Feather Express

dicobrazz, 2019-07-01:
**Describe the bug**
Out of the box nRF52840 Feather Express. Trying to upload example - "rssi_proximity_peripheral". Getting error in Serial Monitor - "Unable to init Bluefruit".

I have, also, tried to upload - "beacon" example. Nothing in the Serial Port and nothing in the debug log, the device is programmed successfully. 

I have then, tried some other libraries - FONA library and examples with FONA808 connected to the board, all works as expected.

**Set up (please complete the following information)**
  - Adafruit Feather nRF52840 Express (revD)
  - Mac OS X 10.14.4
  - Debug Log - SoftDevice's RAM requires: 0x20002530  

**To Reproduce**
Steps to reproduce the behavior:
1. Connect board via standard USB port (purchased suggested with board at Adafruit website)
2. Initial setup
3. Upload example - "rssi_proximity_peripheral"
4. Error - "Unable to init Bluefruit"

**Expected behavior**
After the example has been uploaded, the first thing I should've have seen in the Serial Monitor - "Bluefruit initialized (peripheral mode)"

**Screenshots**
[image]

dicobrazz, 2019-07-01:
I have also tried to call:

`InternalFS.format();`

at beginning of void setup() and no luck so far.
[image]

dicobrazz, 2019-07-01:
Resolved, a mistype in example code:

RSSI example:
at [line 63](https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/libraries/Bluefruit52Lib/examples/Projects/rssi_proximity/rssi_proximity_peripheral/rssi_proximity_peripheral.ino#L63) added !

beacon example:
at [line 58](https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/libraries/Bluefruit52Lib/examples/Peripheral/beacon/beacon.ino#L58) commented out call for suspend, probably needs to be moved or maybe its ok, but for me worked when disabling it.

closed 2019-07-01 by dicobrazz

hathach, 2019-07-02:
thanks for issue and the pr :+1:

dicobrazz, 2019-07-03:
> thanks for your issue and pr 👍

happy to help!