adafruit/Adafruit_nRF52_Arduino #305: BLEUART write description error [Bug]

toufou54, 2019-07-16:
**Describe the bug**
UART not working after bootloader and lib updates.
Trying with examples sketches and BlueFruit Android App ornordic app:
**"error write description"** when trying UART connection

**Set up (please complete the following information)**
  - nRF52832 Feather
  - windows 7 / arduino 1.8.9
  - Fresh upgrade to 0.2.11 bootloader

**To Reproduce**
- update bootloader to the last
- update to last librairies
- writ peripherical uart exemple sketch
- open Bluefruit android app
- connect to device 
- try UART mode.. fail

**Expected behavior**
- UART communication success

Splitframe, 2019-07-16:
Can you set the Debug Level to 2 and post the serial output?
(Arduino IDE: Tools -> Debug, Platform.io: add "build_flags = -DCFG_DEBUG = 2" to platform.ini )

toufou54, 2019-07-16:
Hi, thanks for your help, I've try downgrading but unfortunately does not work any more.
here the serial output, but I'm not sure debug level is working, I've selected Tools -> Debug level 2 (full) 
But not see any changes, no more than usual messages ....
The android app says "Uart protocol can not be initialized"

Bluefruit52 BLEUART Example
---------------------------

[CFG   ] SoftDevice's RAM requires: 0x20002FA0
Please use Adafruit's Bluefruit LE app to connect in UART mode
Once connected, enter character(s) that you wish to send

BSP Library : 0.9.1
Bootloader  : s132 6.1.1 r0
Serial No   : B6C0ADDBCAE9F832

--------- SoftDevice Config ---------
Max UUID128     : 10
ATTR Table Size : 2048
Service Changed : 1
Peripheral Connect Setting
  - Max MTU         : 247
  - Event Length    : 6
  - HVN Queue Size  : 3
  - WrCmd Queue Size: 1

--------- BLE Settings ---------
Name            : Bluefruit52
Max Connections : Peripheral = 1, Central = 0 
Address         : EE:7E:43:45:A6:4B (Static)
TX Power        : 4 dBm
Conn Intervals  : min = 20.00 ms, max = 30.00 ms
Conn Timeout    : 2000.00 ms
Peripheral Paired Devices: 

[BLE   ] BLE_GAP_EVT_CONNECTED : Conn Handle = 0
[GAP   ] Conn Interval= 45.000000
[BLE   ] BLE_GATTC_EVT_HVX : Conn Handle = 0
[BLE   ] BLE_GATTC_EVT_CHAR_VAL_BY_UUID_READ_RSP : Conn Handle = 0
Connected to OnePlus 3T
[BLE   ] BLE_GAP_EVT_CONN_PARAM_UPDATE : Conn Handle = 0
[GAP   ] Conn Interval= 30.000000
[BLE   ] BLE_GAP_EVT_DATA_LENGTH_UPDATE_REQUEST : Conn Handle = 0
[GAP   ] Data Length Req is (tx, rx) octets = (123, 123), (tx, rx) time = (1096, 1096) us
[BLE   ] BLE_GAP_EVT_DATA_LENGTH_UPDATE : Conn Handle = 0
[GAP   ] Data Length is (tx, rx) octets = (123, 123), (tx, rx) time = (1096, 1096) us
[BLE   ] BLE_GAP_EVT_DISCONNECTED : Conn Handle = 0
[GAP   ] Disconnect Reason 0x13

Disconnected

toufou54, 2019-08-09:
Any help ? Please

hathach, 2019-08-10:
Sorry, I am too busy with other issues just now.

aricooperdavis, 2022-11-28:
I am experincing this bug too, happy to help debug if possible!

aricooperdavis, 2022-11-28:
For other experiencing behaviour like this, try "forgetting" or "unpairing" the device in your phone's bluetooth menu - this problem can be caused by your phone cacheing the feather's capabilities, which may exclude uart.