adafruit/Adafruit_nRF52_Arduino #306: [Bug] Serial.read returns wrong value when no characters available

pyro9, 2019-07-19:
According to documentation, Serial.read() returns an int value of the character read or -1 if none available.

In fact, when none are available, it returns 0xff which is also a plausible character for it to read.

Additionally, setTimeout is not honored.

Tested on Bluefruit nrf52840.

Run the following:
void setup() {
  Serial.begin(115200);
  Serial.setTimeout(10000);
}

void loop() {
  int c;

  c = Serial.read();
  Serial.printf("c=%d\n",c);
  delay(1000);
}

Open serial console. It should report a value of -1 every 11 seconds. In fact, it reports 255 every second.

closed 2019-07-19 by pyro9

dhalbert, 2019-07-19:
@pyro9 Why did you close this?

pyro9, 2019-07-19:
I realized my BSP was not up to date and the new version had some revisions to relevant code. However, I will be opening a revised bug report since the bad behavior seems to have moved rather than going away in 0.11.1.

pyro9, 2019-07-19:
See issue #307 BUG Serial is not 8 bit clean