adafruit/Adafruit_nRF52_Arduino #307: [Bug] Serial is not 8 bit clean

pyro9, 2019-07-19:
Tested on nrf52840 Bluefruit Feather Express rev. D with BSP 0.11.1.

Reading any byte from Serial where the MSB is set is mis-interpreted as no data available.
Adafruit_TinyUSB_Core/Adafruit_USBD_CDC.cpp read returns the value of tud_cdc_read_char() promoted from signed char to signed int.

Since Stream::readBytes calls read, it fails to read the full count when a byte has MSB set.

The following quick and dirty hack correctly reads the full count:

    size_t readBytes(Stream &s, uint8_t *buf, size_t count) {
      uint8_t *pos = buf;

      for(int i = count; i; i--) {
          while(!s.available());
          *pos = s.read();
          pos++;
      }
      return count;
    }

(of course, it doesn't honor timeout since it's a quick and dirty hack)

To get what appears to be the intended behavior throughout the stack, it looks like at least tud_cdc_n_peek and tud_cdc_n_read_char need to return signed int so they can express 0-0xff or -1 correctly.

pyro9, 2019-07-31:
Patch applied

closed 2019-07-31 by pyro9

hathach, 2019-07-31:
Thanks @pyro9 for finding and fixing this bug