adafruit/Adafruit_nRF52_Arduino #310: [Bug] nrf52832 cannot parseReportByType of other nrf52832

Splitframe, 2019-07-24:
**Describe the bug**
When trying to connect two nrf52832 the central one cannot parse the report
of the peripheral in the ScanCallback.

**Set up (please complete the following information)**
BSP Library : 0.10.1
Bootloader  : s132 6.1.1
(For Both)

**To Reproduce**
Steps to reproduce the behavior:
On Central:
```
void centScanCallback(ble_gap_evt_adv_report_t* report) {
  uint8_t buffer[32];
  if(Bluefruit.Scanner.parseReportByType(report, BLE_GAP_AD_TYPE_COMPLETE_LOCAL_NAME, buffer, sizeof(buffer))) {
    log("Success");
    // Decide to connect depending on name in buffer.
  } else {
    log("Failure");
  }
}
```
On Peripheral:
```
void startAdvertisement(char advertisementName[]) {
  while(Bluefruit.Advertising.isRunning())
    Bluefruit.Advertising.stop();

  Bluefruit.Advertising.clearData();
  Bluefruit.setName(advertisementName);
  Bluefruit.Advertising.addFlags(BLE_GAP_ADV_FLAGS_LE_ONLY_GENERAL_DISC_MODE);
  Bluefruit.Advertising.addTxPower();
  Bluefruit.Advertising.addService(prphUART);
  Bluefruit.Advertising.setInterval(32, 244);

  Bluefruit.ScanResponse.addName();

  Bluefruit.Advertising.start();
}
```

**Expected behavior**
The name should be parsed into the buffer.

Splitframe, 2019-07-26:
Additional Information:
In the central setup:
```
  Bluefruit.begin(1, 1);
  Bluefruit.setName("Server Node");
  Bluefruit.setTxPower(4);
  Bluefruit.configPrphBandwidth(BANDWIDTH_MAX);
  Bluefruit.configCentralBandwidth(BANDWIDTH_MAX);

  // Central Scan setup
  Bluefruit.Scanner.restartOnDisconnect(false);
  Bluefruit.Scanner.setInterval(160, 80); // in unit of 0.625 ms
  Bluefruit.Scanner.filterUuid(prphUART.uuid);
  Bluefruit.Scanner.useActiveScan(false);
  Bluefruit.Scanner.filterRssi(-70);
```

closed 2019-07-26 by Splitframe

hathach, 2019-07-26:
@Splitframe did you figure out the reason and fixed it

Splitframe, 2019-07-26:
It seems to have been a combination of an uncleared ScanResponse and active scan disabled.
I wonder why the Bluetooth app on my phone worked then though...

asainrath, 2020-04-04:
Device: nrf52840
I still see this issue in version 18.5 too.
Sometime it parse the name to buffer but not always.

if I enable, Bluefruit.Scanner.filterUuid(), it doesn't parse the name to buffer at all. 

Error can be reproduced from the following code.
-------------------------------------------------------------------------------------
#include <bluefruit.h>

const uint8_t CUSTOM_UUID[] =
{
    0xA0, 0xDB, 0xD3, 0x6A, 0x00, 0xA6, 0xF7, 0x8C,
    0xE7, 0x11, 0x8F, 0x71, 0x1A, 0xFF, 0x67, 0xDF
};

BLEUuid test_uuid = BLEUuid(CUSTOM_UUID);

void setup()
{
  Serial.begin(115200);
  while ( !Serial ) delay(10);   
  Bluefruit.begin(1, 1);
  Bluefruit.setTxPower(4);

  Bluefruit.Scanner.setRxCallback(scan_callback);
  Bluefruit.Scanner.restartOnDisconnect(true);
  Bluefruit.Scanner.filterRssi(-60);         
//  Bluefruit.Scanner.filterUuid(test_uuid);            
  Bluefruit.Scanner.setInterval(160, 80);    
  Bluefruit.Scanner.useActiveScan(true);      
  Bluefruit.Scanner.start(0);             
  Serial.println("Scanning ...");
}

void scan_callback(ble_gap_evt_adv_report_t* report)
{
  uint8_t buffer[32];
  memset(buffer, 0, sizeof(buffer));

  Serial.printBufferReverse(report->peer_addr.addr, 6, ':');
  Serial.print("  ");
  
     if(Bluefruit.Scanner.parseReportByType(report, BLE_GAP_AD_TYPE_COMPLETE_LOCAL_NAME, buffer, sizeof(buffer)))
  {
    Serial.printf("%14s",buffer);
    memset(buffer, 0, sizeof(buffer));
  }
    else if (Bluefruit.Scanner.parseReportByType(report, BLE_GAP_AD_TYPE_SHORT_LOCAL_NAME, buffer, sizeof(buffer)))
  {
    Serial.printf("%14s",buffer);
    memset(buffer, 0, sizeof(buffer));
  }
  else
  {
   Serial.print("Failed to read the device name"); 
  }
  
  Serial.println();
  Bluefruit.Scanner.resume();
}

void loop() 
{
 
}

hathach, 2020-04-05:
can you also post the serial output when debug level =2 is eanbled, keep it simple with the log that matters when issue occurs.