adafruit/Adafruit_nRF52_Arduino #314: InternalFileSystem Examples no Longer Compile

derknorton, 2019-08-01:
The [examples](https://github.com/adafruit/Adafruit_nRF52_Arduino/tree/master/libraries/InternalFileSytem/examples) in `InternalFileSystem` no longer compile.  They need to be updated, the [bonding source code](https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/libraries/Bluefruit52Lib/src/utility/bonding.cpp) is a good place to see the current file system call syntax.

hathach, 2019-08-05:
Please provide your information on OS, Arduino IDE, BSP version, board variant etc... It still compiles fine here.

derknorton, 2019-08-05:
Thanks @hathach for the quick response.  Here are my specs:
OS: Mac OSX 10.14.5
IDE: Arduino 1.8.9
Board: Adafruit Bluefruit Feather nRF52832
Bootloader: 0.2.11 SoftDevice s132 6.1.1

I was getting compiler errors when attempting to open a file as follows:
```
    File file(InternalFS);
    file.open(FILENAME, FILE_O_READ);    <= the compiler complained about this line
    file.read(buffer, BUFFER_SIZE);
    file.close();
```
I switched it to be like the bonding source and it worked:
```
    File file(FILENAME, FILE_O_READ, InternalFS);
    file.read(buffer, BUFFER_SIZE);
    file.close();
```
So I thought maybe the interface had changed after the example code was written.  I just switched it back in my code and it seems to work either way.  So it must have been pilot error, you can close this issue.  Thanks.

closed 2019-08-05 by derknorton