adafruit/Adafruit_nRF52_Arduino #318: [Feature] Maximize bandwidth

hathach, 2019-08-06:
https://www.novelbits.io/bluetooth-5-speed-maximum-throughput/

Splitframe, 2019-08-27:
I achieved 172Kbyte/s with:

`configPrphConn(247, 150, 3, BLE_GATTC_WRITE_CMD_TX_QUEUE_SIZE_DEFAULT);`  
and  
`configCentralConn(247, 150, 3, BLE_GATTC_WRITE_CMD_TX_QUEUE_SIZE_DEFAULT);`  

in the Bluefruit.cpp as well as:

```
Bluefruit.Periph.setConnInterval(135, 165);
conn->requestConnectionParameter(150); // in unit of 1.25
```

In the peripheral sketch.

I bet you can even tweak it further, but this setting was the most reliable for our use case
and fast enough. A big thanks to @hathach for implementing the DLE and PHY negotiation.

hathach, 2019-08-28:
172 KB/s is pretty fast, I am planning to spend more time testing `throughput.ino` and `central_throughput.ino` with different config and result. Thanks for your feedback :)