adafruit/Adafruit_nRF52_Arduino #319: [Bug] Sometimes AppID and AppName comes empty

trx222, 2019-08-07:
**Describe the bug**
From time to time incomming notifications do not show App-ID and App-Name. Message, Title and category are shown in terminal.

**Set up (please complete the following information)**
  - Bluefruit nRF52832 feather with latest BSP version (0.12.0)
  - MacOS

**To Reproduce**
Steps to reproduce the behavior:
1. Start Example ancs with open terminal
2. connect IphoneX via bluetooth to Bluefruit52
3. wait for incomming notifications shown in terminal

**Expected behavior**
all notifications should have AppID and AppName ... other bluetooth devices show always an app name

hathach, 2020-03-19:
sorry for the long wait, I somehow always find myself busy with other overdued work. 0.12.0 is a bit old, can you try again with the current `0.19.0`. Please also post your serial log, please also point out which is the missing appID, appName (by using your iPhone), some builtin app like Screen Time doens't seems to have an appName.

trx222, 2020-03-19:
I will try a newer version. What I can already say is that the notifications are for normal apps ... I see this at the text which is sometimes there ... and the next push works fine then. 
I will check it next week and give you detailed log them. thanks a lot.

hathach, 2020-03-20:
@trx222 I made some changes to the ancs example, to only get relevant attribute. and chop off the title Unicode. You may want to copy the `ancs_notification_callback()`  from it to replace current one (whole sketch won't work due to the new API).

https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/develop/libraries/Bluefruit52Lib/examples/Peripheral/ancs/ancs.ino#L178

I did an simple test yesterday, `com.apple.ScreenTimeNotifications` don't have appName, you got to map it to **Screen Time** I guess.