adafruit/Adafruit_nRF52_Arduino #321: [Bug] Included RotaryEncoder library allows invalid modification of SAMPLEPER

henrygab, 2019-08-11:
I can write the fix, and submit a PR.  However, is this the right place to do so, or is this library living elsewhere?

**Describe the bug**
Per nRF52840 product specifications:

> If the SAMPLEPER value needs to be changed, the QDEC shall be stopped using the STOP task. SAMPLEPER can be then changed upon receiving the STOPPED event, and QDEC can be restarted using the START task. Failing to do so may result in unpredictable behaviour

However, the library does not ensure the peripheral is stopped prior to changing the sample period:
https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/64b555531a7c6815a5c6398fabddf1f666f6b70d/libraries/RotaryEncoder/RotaryEncoder.cpp#L114-L117

**Set up (please complete the following information)**
  - nRF52840, BSP 0.12.0
  - N/A (but Windows 10 x64)
  - Serial debug log == N/A

**To Reproduce**
Steps to reproduce the behavior:
1. In sketch, enable and use the hardware rotary encoder.
2. while the rotary encoder is started, change sample period using the above library API.

**Expected behavior**
The library should do the following sequence:
1. Save current peripheral state to local variables (enabled, running):
2. If peripheral is enabled and running:
  a. Send STOP task to the QDEC peripheral
  b. Wait for STOPPED event from QDEC peripheral
3. Set the new sample rate
4. IF peripheral was enabled and running at start of function:
  a. Send START task to the QDEC peripheral

**Screenshots**
N/A

**Additional context**
The following sentence in any chip's product specification is a situation to be avoided:

> Failing to do so may result in unpredictable behaviour

henrygab, 2019-08-11:
It's easy enough to know if the QDEC peripheral is enabled.
**_Open questions:_**
1. Is it possible to reliably detect if QDEC peripheral is started?
2. If QDEC is not started, and TASKS_STOP is set, will it set EVENTS_STOPPED?

henrygab, 2019-08-12:
The answers to my own questions:
1. **_No_**. There does not appear to be any way to determine if the peripheral is started.
2. **_No_**. If the peripheral is stopped, setting TASKS_STOP will **_not_** cause EVENTS_STOPPED.

It does appear that it's safe to call start() repeatedly.
It also appears that EVENTS_STOPPED occurs immediately (or, within register read delay, which is <=4 cycles) if the peripheral was stopped … at least if LED is not enabled and in use.

henrygab, 2019-08-12:
A SystemView trace shows the current library generating an interrupt for each and every sample, even when there is zero change.  This is a HUGE amount of processing overhead (INT -> ISR -> callback -> … only to see diff of zero … for every sample).

The whole point of having the hardware QDEC is to only trigger when a change is detected.  The way the library currently stands, enabling reporting does **_not_** disable the interrupts for every sample, thus simply increasing wasted CPU cycles.

Would you be open to the following changes for this library, provided via PR?

Changes are split into two parts: Debug improvements and Functional changes.  Listing each in turn:

#### Debug improvements
Changes when CFG_DEBUG>=3 (Segger SystemView enabled build):
1. Enable events for ISR entry/exit, using `SEGGER_SYSVIEW_RecordEnterISR()` and `SEGGER_SYSVIEW_RecordExitISR()`.
2. Add ad-hoc logging with levels (CRIT, ERR, WARN, INFO, VERBOSE), using `SEGGER_SYSVIEW_ErrorfHost()` / `SEGGER_SYSVIEW_WarnfHost()` / `SEGGER_SYSVIEW_PrintfHost()`

The debug improvements would compile to nothing via pre-processor macro expansion, and would be fully removed in builds without SystemView enabled.

#### Functional changes
1. Default to `setReporter(true)`
2. Disable all shortcuts (`REPORTRDY -> READCLRACC` was previously enabled with registered callback ... but I don't think it did what was expected)
3. Default to only use `EVENTS_REPORTRDY` when callback is registers (no callback on zero change) 

#### Request
Please let me know if the above changes are of interest.  I have already written the majority of the code, both to understand the perf issues listed above (_an ISR and callback for every sample_), and for use in a project.

I'd prefer these fixes get shared back with the community.  Note that the code would very likely work on all the nRF52 boards.  I can test build against all the platforms, although I can only test real hardware with the nRF52840 at present.

hathach, 2019-08-20:
@henrygab Please give me a bit more time, I am currently in middle of something else. Will check this out next week or so.

henrygab, 2019-08-20:
Hi @hathach, OK, no rush.

I have a [branch](https://github.com/henrygab/Adafruit_nRF52_Arduino/tree/RotaryLib) that does the above.  While I'm not done testing, it does seem solid already.  I'm still thinking about a couple things.... 

<details>
  <summary>Should `read()` update value if callback set?</summary>

  <p>

I'm not sure it makes sense, when a callback is registered, for the `read()` function to read the accumulator register.  The reason is that, if a sketch has registered a callback function, it is reasonable for it to understand that to mean that the callback function will be used for all updates.

However, where the caller also calls read(), it's possible that a non-zero value in the accumulator register exists, thus removing those extra values from the registered callback function.  It's also possible that a value would be double-counted:
1. `read()` function:
    A. Sets TASKS_READCLRACC
    B. unknown value is stored in ACCREAD
    C. but interrupt fires before ACCREAD value is read
2. ISR:
    A. Sets TASKS_READCLRACC
    B. New value is stored in ACCREAD
    C. ISR reads new value from ACCREAD and acts on it
    D. ISR exits normally
3. `read()` function resumes:
    D. New value is read from ACCREAD

_Why could this be bad?_
As an example, consider if the callback is also calculating velocity / acceleration.  Losing transitions would make it seems that the turn is going slower than it should.  Or, if the encoder's cap has a marker for zero ... it would slowly get out-of-true.

I still have to think it through some, but I think I may advocate either:
1. Not updating from `read()` if a callback is set, or
2. Calling the callback from `read()` if a non-zero accumulator value is read

</details>

Expand the above if you want to read the thoughts … It's a bit much, so I wrote it collapsed by default.

henrygab, 2019-09-30:
Closing this because the rotary encoder lib provided is not a great choice, neither when using the nRF52's built-in HW decoder peripheral, nor when creating software-based decoder.

I am looking at creating a state-machine based solution, with MIT license.   When that's tested, I will open a new issue for consideration of inclusion.

closed 2019-09-30 by henrygab