adafruit/Adafruit_nRF52_Arduino #323: [Feature] Add an example of how to do a write() to a BLEClientCharacteristic from a ble Central

dsteinman, 2019-08-15:
Hi, I'm running into a stumbling block and could use some help.  None of the NRF52 Central examples show how to write to a BLEClientCharacteristic, and I'm having trouble figuring out how to do it or even if it can be done:

- https://github.com/adafruit/Adafruit_nRF52_Arduino/tree/master/libraries/Bluefruit52Lib/examples/Central/central_bleuart
- https://github.com/adafruit/Adafruit_nRF52_Arduino/tree/master/libraries/Bluefruit52Lib/examples/Central/central_bleuart_multi
- https://github.com/adafruit/Adafruit_nRF52_Arduino/tree/master/libraries/Bluefruit52Lib/examples/Central/central_custom_hrm
- https://github.com/adafruit/Adafruit_nRF52_Arduino/tree/master/libraries/Bluefruit52Lib/examples/Central/central_hid
- https://github.com/adafruit/Adafruit_nRF52_Arduino/tree/master/libraries/Bluefruit52Lib/examples/Central/central_scan
- https://github.com/adafruit/Adafruit_nRF52_Arduino/tree/master/libraries/Bluefruit52Lib/examples/Central/central_scan_advanced

https://github.com/adafruit/Adafruit_nRF52_Arduino/tree/master/libraries/Bluefruit52Lib/examples/Central/central_ti_sensortag_optical

All of the examples only show how to Scan, display scan information, use the UART, and the "advanced" one show how to receive notifications but not how to do writes.  I know this can be done because I can write to the peripheral using my PC, but I want to do it from an NRF52 feather.

The API looks like it should be easy:

```
uint16_t BLEClientCharacteristic::write(const void* data, uint16_t len)
```

but it's not working for me and I don't know what I'm doing wrong.  Here's a sample of what I'm trying to, tthis is the code on my Central device (a NRF52832 feather) and I want to send data to a peripheral I connected to:

```
BLEClientCharacteristic myCharacterstic = BLEClientCharacteristic(0x123456...);
...
// after connecting
const uint8_t mydata[] = {1,2,3};
myCharacterstic.write(mydata, 4);
```

What exactly am I supposed to send in for void* data ?

Shouldn't the write() method require a conn_handle or something to identify the exact peripheral I want to send data to?

Thanks for any help you can provide!

dsteinman, 2019-08-18:
I was able to figure out what I was doing wrong.  The solution turned out to be simple:

```
const char mydata[] = {1,2,3};
const char* mychardata = mydata;
myCharacterstic.write( (const uint8_t*) mychardata, strlen(mychardata) );
```

I also forgot to perform a ".discover()" before performing the write()'s.  So I unknowingly had 2 problems which made it tougher to diagnose.

Something isn't quite sitting right with me in regards to the BLEClientCharacteristic write() API.  I think there should be an alternate version of the write() method with conn_handle exposed and with a const char[] as the parameter.  That way it would closer match the BLECharacteristic write() and give the ability to send distinct devices distinct messages.

hathach, 2019-08-20:
BLEClientCharacteristic doesn't behave like the BLECharacteristic (server). Here is how it work
- 1 server characteristic can be connected to multiple clients
- however 1 client characteristic can only connect to ONLY ONE server

thus the connection handle parameter only exists in the server API.

dsteinman, 2019-08-20:
Thanks for pointing this out.  I actually thought BLEClientCharacteristic was how a ble server communicates to the clients, I guess my understanding of these classes is the reverse of what it ought to be.  If I change to BLECharacteristic the same question remains though because the BLECharacteristic write() api's also don't expose a `conn_handle`:

[https://learn.adafruit.com/introducing-the-adafruit-nrf52840-feather/blecharacteristic](https://learn.adafruit.com/introducing-the-adafruit-nrf52840-feather/blecharacteristic)

```
/*------------- Write -------------*/
uint16_t write(const void* data, uint16_t len);
uint16_t write(const char* str);
 
uint16_t write8    (uint8_t  num);
uint16_t write16   (uint16_t num);
uint16_t write32   (uint32_t num);
uint16_t write32   (int      num);
```

hathach, 2019-08-20:
BLECharacteristic write() also doesn't requires connection handle because it doesn't generate any BLE transaction, only notify/indicate does. Write with server is simply update its value.

We don't have resource to write example for every API(), but I will be to merge if anyone submit an PR for it.

closed 2019-08-20 by hathach

dsteinman, 2019-08-20:
I appreciate the help and info @hathach, I may try to make such a client/server example.

hathach, 2019-08-21:
If possible please submit your example as PR. I could help to review and troubleshoot if needed. Best to keep it as simple as possible, maybe modify the hrm or hts central/peripheral. I think they also have writable characteristics as well.