adafruit/Adafruit_nRF52_Arduino #325: nRF52832 frequently connect/disconnect occur assertion "head >= 2 && head <= lfs->cfg->block_count"

shaoyuan1943, 2019-08-16:
Hi~
I'm using a nRF52832 feather to build keyboard. I found a strange bug when iPhone disconnecting and reconnecting keyboard frequently can cause error:

10:36:26.458 -> assertion "head >= 2 && head <= lfs->cfg->block_count" failed: file "C:\Users\cheng\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.11.1\libraries\Adafruit_LittleFS\src\littlefs\lfs.c", line 1144, function: lfs_ctz_find
10:36:33.668 -> assertion "head >= 2 && head <= lfs->cfg->block_count" failed: file "C:\Users\cheng\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.11.1\libraries\Adafruit_LittleFS\src\littlefs\lfs.c", line 1144, function: lfs_ctz_find

In my project, iPhone is peripheral and keyboard is cent, iPhone connect keyboard when it started. The connect/disconnect callback code like this:

void prphConnectCallback(uint16_t conn_handle) {
  // Get the reference to current connection
  BLEConnection* connection = Bluefruit.Connection(conn_handle);
  perph_conn_handle = conn_handle;

  // debug information
  char peer_name[32] = { 0 };
  connection->getPeerName(peer_name, sizeof(peer_name));

  LOG("[Prph] Connected handle: %d, name: %s\r\n", conn_handle, peer_name);
}

void prphDisconnectCallback(uint16_t conn_handle, uint8_t reason) {
  (void) conn_handle;
  (void) reason;

  startAdv();

  // debug information
  LOG("[Prph] Disconnected\r\n");
}

The complete serial logs is(Level 0):
02:31:05.162 -> start...
02:31:05.435 -> in low power
02:31:05.468 -> LN-REDOX bluetooth keyboard RHS started
02:31:05.876 -> [SCAN] scanned, start connect...
02:31:06.186 -> [Cent] Connected handle: 0, name: LHS
02:31:10.046 -> [Prph] Connected handle: 1, name: iPhone
02:31:22.517 -> [Prph] Disconnected
02:31:26.375 -> [Prph] Connected handle: 1, name: iPhone (2
02:31:41.828 -> [Prph] Disconnected
02:31:47.139 -> [Prph] Connected handle: 1, name: iPhone (2
02:31:59.740 -> [Prph] Disconnected
02:32:03.654 -> [Prph] Connected handle: 1, name: iPhone (2
02:32:31.155 -> [Prph] Disconnected
02:32:35.173 -> [Prph] Connected handle: 1, name: iPhone (2
02:32:52.566 -> [Prph] Disconnected
02:32:56.388 -> [Prph] Connected handle: 1, name: iPhone (2
02:33:11.480 -> [Prph] Disconnected
02:33:14.658 -> [Prph] Connected handle: 1, name: iPhone (2
02:33:27.269 -> [Prph] Disconnected
02:33:31.146 -> [Prph] Connected handle: 1, name: iPhone (2
02:33:56.366 -> [Prph] Disconnected
02:34:03.249 -> [Prph] Connected handle: 1, name: iPhone (2
02:34:04.756 -> [Prph] Disconnected
02:34:16.321 -> [Prph] Connected handle: 1, name: iPhone (2
02:37:26.456 -> [Prph] Disconnected
02:42:45.820 -> [Prph] Connected handle: 1, name: iPhone (2
02:53:56.825 -> [Prph] Disconnected
08:45:48.984 -> [Prph] Connected handle: 1, name: 
09:03:59.574 -> [Prph] Disconnected
09:33:22.403 -> [Prph] Connected handle: 1, name: iPhone (2
09:34:09.581 -> [Prph] Disconnected
10:03:07.080 -> [Prph] Connected handle: 1, name: iPhone (2
10:03:28.158 -> [Prph] Disconnected
10:23:34.210 -> [Prph] Connected handle: 1, name: 
10:23:34.659 -> [Prph] Disconnected
10:23:42.526 -> [Prph] Connected handle: 1, name: 
10:23:42.526 -> [Prph] Disconnected
10:23:43.323 -> [Prph] Connected handle: 1, name: 
10:23:44.549 -> [Prph] Disconnected
10:23:49.709 -> [Prph] Connected handle: 1, name: 
10:23:51.040 -> [Prph] Disconnected
10:23:53.630 -> [Prph] Connected handle: 1, name: 
10:23:54.343 -> [Prph] Disconnected
10:27:51.664 -> [Prph] Connected handle: 1, name: 
10:27:52.276 -> [Prph] Disconnected
10:27:55.238 -> [Prph] Connected handle: 1, name: 
10:27:55.272 -> [Prph] Disconnected
10:27:56.631 -> [Prph] Connected handle: 1, name: iPhone (2
10:28:13.872 -> [Prph] Disconnected
10:28:26.004 -> [Prph] Connected handle: 1, name: iPhone (2
10:28:27.223 -> [Prph] Disconnected
10:28:32.137 -> [Prph] Connected handle: 1, name: iPhone (2
10:28:33.885 -> [Prph] Disconnected
10:36:19.230 -> [Prph] Connected handle: 1, name: iPhone (2
10:36:26.458 -> assertion "head >= 2 && head <= lfs->cfg->block_count" failed: file "C:\Users\cheng\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.11.1\libraries\Adafruit_LittleFS\src\littlefs\lfs.c", line 1144, function: lfs_ctz_find
10:36:33.668 -> assertion "head >= 2 && head <= lfs->cfg->block_count" failed: file "C:\Users\cheng\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.11.1\libraries\Adafruit_LittleFS\src\littlefs\lfs.c", line 1144, function: lfs_ctz_find

I want to know when this error will occur and how to solve it.

bootloader:
feather_nrf52832_bootloader-0.2.11_s132_6.1.1

bsp:
0.11.1

Plz~

hathach, 2019-08-16:
Could you provide all the information as suggested by the issue template, your simplified sketch and exact steps to reproduce the issue. Currently I don't have time but it may help other to troubleshoot this

shaoyuan1943, 2019-08-17:
@hathach ok，I will give you level 1 or level 2 logs later and more test information

shaoyuan1943, 2019-08-17:
@hathach Hi~
Hi, hatach
Today I got this error again, the level 2 log like this:
......
11:07:32.648 -> [BOND  ] Loaded keys from file /adafruit/bond_prph/4424
11:07:32.820 -> [BLE   ] BLE_GAP_EVT_CONN_SEC_UPDATE : Conn Handle = 1
11:07:32.820 -> [BOND  ] Loaded CCCD from file /adafruit/bond_prph/4424 ( offset = 102, len = 50 bytes )
11:07:32.820 -> [CHR   ] Notify 8 bytes
11:07:32.854 -> [BLE   ] BLE_GATTS_EVT_HVN_TX_COMPLETE : Conn Handle = 1
11:07:32.889 -> [BLE   ] BLE_GAP_EVT_CONN_PARAM_UPDATE : Conn Handle = 1
11:07:32.889 -> [GAP   ] Conn Interval= 30.000000
11:07:32.889 -> [CHR   ] Notify 8 bytes
11:07:32.923 -> [BLE   ] BLE_GATTS_EVT_HVN_TX_COMPLETE : Conn Handle = 1
11:07:32.923 -> [CHR   ] Notify 8 bytes
11:07:32.992 -> [SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
11:07:33.126 -> bool BLEAdvertising::stop(): 391: verify failed, error = NRF_ERROR_INVALID_STATE
11:07:33.126 -> bool BLEAdvertising::_start(uint16_t, uint16_t): 366: verify failed, error = NRF_ERROR_CONN_COUNT
11:07:33.126 -> [Prph] Disconnected
11:07:33.333 -> [BLE   ] BLE_GAP_EVT_PHY_UPDATE_REQUEST : Conn Handle = 1
11:07:33.333 -> [GAP   ] PHY request tx: 2 Mbps, rx: 2 Mbps
11:07:33.333 -> [BLE   ] BLE_GATTS_EVT_HVN_TX_COMPLETE : Conn Handle = 1
11:07:33.333 -> [CHR   ] Notify 8 bytes
11:07:33.367 -> [BLE   ] BLE_GATTC_EVT_CHAR_VAL_BY_UUID_READ_RSP : Conn Handle = 1
11:07:33.367 -> [Prph] Connected handle: 1, name:  iPhone (2
11:07:33.433 -> [BLE   ] BLE_GATTS_EVT_HVN_TX_COMPLETE : Conn Handle = 1
11:07:33.433 -> [CHR   ] Notify 8 bytes
11:07:33.502 -> [BLE   ] BLE_GATTS_EVT_HVN_TX_COMPLETE : Conn Handle = 1
11:07:33.502 -> [CHR   ] Notify 8 bytes
11:07:33.570 -> [BLE   ] BLE_GATTS_EVT_HVN_TX_COMPLETE : Conn Handle = 1
11:07:33.570 -> [CHR   ] Notify 8 bytes
11:07:33.605 -> [BLE   ] BLE_GATTS_EVT_HVN_TX_COMPLETE : Conn Handle = 1
11:07:33.605 -> [CHR   ] Notify 8 bytes
11:07:33.673 -> [BLE   ] BLE_GAP_EVT_PHY_UPDATE : Conn Handle = 1
11:07:33.673 -> [GAP   ] PHY active tx: 2 Mbps, rx: 2 Mbps
11:07:33.673 -> [BLE   ] BLE_GATTS_EVT_HVN_TX_COMPLETE : Conn Handle = 1
11:07:33.673 -> [CHR   ] Notify 8 bytes
11:07:33.774 -> [BLE   ] BLE_GAP_EVT_DATA_LENGTH_UPDATE_REQUEST : Conn Handle = 1
11:07:33.774 -> [GAP   ] Data Length Req is (tx, rx) octets = (251, 251), (tx, rx) time = (2120, 2120) us
11:07:33.808 -> [BLE   ] BLE_GAP_EVT_DATA_LENGTH_UPDATE : Conn Handle = 1
11:07:33.808 -> [GAP   ] Data Length is (tx, rx) octets = (27, 27), (tx, rx) time = (1364, 1364) us
11:07:33.843 -> [BLE   ] BLE_GATTS_EVT_HVN_TX_COMPLETE : Conn Handle = 1
11:07:33.843 -> [CHR   ] Notify 8 bytes
11:07:33.980 -> [BLE   ] BLE_GATTS_EVT_EXCHANGE_MTU_REQUEST : Conn Handle = 1
11:07:33.980 -> [GAP   ] ATT MTU is changed to 23
11:07:34.014 -> [BLE   ] BLE_GATTS_EVT_HVN_TX_COMPLETE : Conn Handle = 1
11:07:34.014 -> [CHR   ] Notify 8 bytes
11:07:34.118 -> [BLE   ] BLE_GATTS_EVT_WRITE : Conn Handle = 1
11:07:34.256 -> [BLE   ] BLE_GATTS_EVT_HVN_TX_COMPLETE : Conn Handle = 1
11:07:34.256 -> [CHR   ] Notify 8 bytes
11:07:34.326 -> [BLE   ] BLE_GATTS_EVT_HVN_TX_COMPLETE : Conn Handle = 1
11:07:34.326 -> [BLE   ] BLE_GATTS_EVT_WRITE : Conn Handle = 1
11:07:34.360 -> 000:  01 00                                           | ..
11:07:34.360 -> 
11:07:34.360 -> [GATTS ] attr's cccd
11:07:34.360 -> [BOND  ] Saved CCCD setting to file /adafruit/bond_prph/4424 ( offset = 102, len = 50 bytes )
11:07:34.360 -> [CHR   ] Notify 8 bytes
11:07:34.430 -> [BLE   ] BLE_GATTS_EVT_WRITE : Conn Handle = 1
11:07:34.430 -> 000:  01 00                                           | ..
11:07:34.430 -> 
11:07:34.430 -> [GATTS ] attr's cccd
11:07:34.430 -> [SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
11:07:34.532 -> [BLE   ] BLE_GATTS_EVT_HVN_TX_COMPLETE : Conn Handle = 1
11:07:34.567 -> [BLE   ] BLE_GAP_EVT_CONN_PARAM_UPDATE : Conn Handle = 1
11:07:34.567 -> [GAP   ] Conn Interval= 15.000000
11:07:34.567 -> [C[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
11:07:34.636 -> OC [SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
11:07:34.636 ->   ] NRF_EVT_FLASH_O[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
11:07:35.009 -> [BOND  ] Saved CCCD setting to file /adafruit/bond_prph/4424 ( offset = 102, len = 50 bytes )
11:07:35.044 -> [SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
11:07:35.282 -> [BLE   ] BLE_GATTS_EVT_WRITE : Conn Handle = 1
[... 167 lines cut ...]

void startAdv(void) {
  Bluefruit.Advertising.addFlags(BLE_GAP_ADV_FLAGS_LE_ONLY_GENERAL_DISC_MODE);
  Bluefruit.Advertising.addTxPower();
  Bluefruit.Advertising.addAppearance(BLE_APPEARANCE_HID_KEYBOARD);
  Bluefruit.Advertising.addService(Hid);

  Bluefruit.ScanResponse.addService(Battery);

  Bluefruit.Advertising.addName();
  Bluefruit.Advertising.restartOnDisconnect(true);
  // in unit of 0.625 ms
  Bluefruit.Advertising.setInterval(32, 244);  
  // number of seconds in fast mode  
  Bluefruit.Advertising.setFastTimeout(30);
  // 0 = Don't stop advertising after n seconds      
  Bluefruit.Advertising.start(0);                
}

void loop() {
// do something

  waitForEvent();
}
```

shaoyuan1943, 2019-08-20:
Hi, @hathach 
How is the progress of this problem? Need other information?

hathach, 2019-08-21:
Look like it is LFS related issue #227 . I won't have time to run your sketch and reproduce this for probably several weeks or more. Please be patient, and try to investigate it. Keep posting your finding here, maybe other with similar issue can come in and help.

shaoyuan1943, 2019-08-21:
OK, thanks for your reply. I will continue to pay attention to this issue and keep posting new informations.

henrygab, 2019-10-05:
Hi @shaoyuan1943 ,

The following output from your debug log is of interest:
> 11:07:35.692 -> [SOC ] NRF_EVT_FLASH_OPERATION_ERROR

I will provide my initial look in a series of responses.  I've collapsed the details, to keep the thread somewhat readable, but simply click to expand for details:

<details>
<summary>InternalFileSystem completely ignores flash errors</summary>

<P>

The above log line indicates some flash operation (likely a write) failed.  The code that outputs that line is in the Bluefruit52Lib library:
https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/8dbe7d950363dc17ed6c571be8dd66f53507feca/libraries/Bluefruit52Lib/src/bluefruit.cpp#L686-L689 … https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/8dbe7d950363dc17ed6c571be8dd66f53507feca/libraries/Bluefruit52Lib/src/bluefruit.cpp#L708-L712

The Bluefruit52Lib library defines flash_nrf5x_event_cb with the 'weak' attribute, allowing a strong definition of the function elsewhere to override the default:
https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/8dbe7d950363dc17ed6c571be8dd66f53507feca/libraries/Bluefruit52Lib/src/bluefruit.cpp#L88-L91

This callback is defined strongly in the InternalFileSystem library:
https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/8dbe7d950363dc17ed6c571be8dd66f53507feca/libraries/InternalFileSytem/src/flash/flash_nrf5x.c#L37-L41

However, as you can see, the function is **_entirely ignoring write failures_**.

</details>

Next up, a look at how the flash is written to...

henrygab, 2019-10-05:
## Nordic API for writing to flash

Here are some rules for using the Nordic API, based on their own header file documentation.  Note that the Adafruit Arduino BSP for the nRF52 always enables the softdevice, even if the radio features are not used, so these rules presume the softdevice is enabled.

<details>
<summary>1. Nordic provides the API that does the actual write to flash as `sd_flash_write()`</summary>

<P>

https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/8dbe7d950363dc17ed6c571be8dd66f53507feca/cores/nRF5/nordic/softdevice/s140_nrf52_6.1.1_API/include/nrf_soc.h#L913-L948

</details>

<details>
<summary>2. <B><I>Exactly one</B></I> write operation may be pending on the flash device at a time:</summary>

<P>

https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/8dbe7d950363dc17ed6c571be8dd66f53507feca/cores/nRF5/nordic/softdevice/s140_nrf52_6.1.1_API/include/nrf_soc.h#L943

</details>

<details>
<summary>3. sd_flash_write() is <B><I>asynchronous</B></I></summary>

<P>

This means a return value of success only means that the write was accepted / queued, not that the write succeeded.

https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/8dbe7d950363dc17ed6c571be8dd66f53507feca/cores/nRF5/nordic/softdevice/s140_nrf52_6.1.1_API/include/nrf_soc.h#L946
</details>

<details>
<summary>4. Must handle NRF_ERROR_BUSY response by retrying the operation</summary>

<P>

OK, technically it's not required to retry, but the write would not be successful.  Retry just makes sense, as this does not indicate an error writing to that area of flash.

https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/8dbe7d950363dc17ed6c571be8dd66f53507feca/cores/nRF5/nordic/softdevice/s140_nrf52_6.1.1_API/include/nrf_soc.h#L943
</details>

<details>
<summary>5. On NRF_SUCCESS, buffer cannot be modified until event NRF_EVT_FLASH_OPERATION_*</summary>

<P>

The buffer used for the write **_must remain unmodified_** until the write completes.  Otherwise, the software is just asking for corrupt data.

https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/8dbe7d950363dc17ed6c571be8dd66f53507feca/cores/nRF5/nordic/softdevice/s140_nrf52_6.1.1_API/include/nrf_soc.h#L931-L932
</details>

henrygab, 2019-10-05:
How the Adafruit InternalFileSystem library writes to flash

<details>
<summary>write function is defined to be _internal_flash_prog</summary>

<P>

See line 106.  `Prog` in this case means 'program the flash memory', aka write.
https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/8dbe7d950363dc17ed6c571be8dd66f53507feca/libraries/InternalFileSytem/src/InternalFileSystem.cpp#L101-L120
</details>

<details>
<summary>_internal_flash_prog() calls flash_nrf5x_write()</summary>

<P>

It just adds an offset so that callers "see" the space as starting at address zero, while supporting different chipsets with different start address for the flash memory.
https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/8dbe7d950363dc17ed6c571be8dd66f53507feca/libraries/InternalFileSytem/src/InternalFileSystem.cpp#L56-L68
https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/8dbe7d950363dc17ed6c571be8dd66f53507feca/libraries/InternalFileSytem/src/InternalFileSystem.cpp#L28-L32
</details>

<details>
<summary>flash_nrf5x_write() calls flash_cache_write()</summary>

<P>

https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/8dbe7d950363dc17ed6c571be8dd66f53507feca/libraries/InternalFileSytem/src/flash/flash_nrf5x.c#L68-L72
</details>

<details>
<summary>flash_cache_write() calls flash_cache_flush() to write previous cache page</summary>

<P>

A small aside.  The cache layer caches all writes until the next write to a different page is requested.  This is great for performance.  However, it means that callers of this function cannot "see" if the write actually made it to the flash or not.
https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/8dbe7d950363dc17ed6c571be8dd66f53507feca/libraries/InternalFileSytem/src/flash/flash_cache.c#L44-L61
</details>

<details>
<summary>flash_cache_flush() calls fc->erase() and fc->program()</summary>

<P>

First, it checks if the contents were actually altered, if there's a verification function.  Else, it first calls `erase()` to ensure the page can be written, and then it calls `program()` to actually write the page with the (modified) data.
</details>

<details>
<summary>fc->program() is fal_program()</summary>

<P>

https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/8dbe7d950363dc17ed6c571be8dd66f53507feca/libraries/InternalFileSytem/src/flash/flash_nrf5x.c#L51-L58
</details>

<details>
<summary>fal_program() has much of the logic</summary>

<P>

https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/8dbe7d950363dc17ed6c571be8dd66f53507feca/libraries/InternalFileSytem/src/flash/flash_nrf5x.c#L115-L151
</details>

henrygab, 2019-10-05:
## Does Adafruit InternalFileSystem library follow the Nordic rules for reliably writing?

The rules (listed above) were:
1. Exactly one write operation may be pending on the flash device at a time:
2. Must handle NRF_ERROR_BUSY response by retrying the operation
3. sd_flash_write() is asynchronous when softdevice is enabled, but synchronous otherwise
4. On NRF_SUCCESS, when softdevice is enabled, the buffer cannot be modified until event NRF_EVT_FLASH_OPERATION_*

First, it's clear that this code handles NRF_ERROR_BUSY, and retries the operation after a delay.  Thus, at least rules 1 and 2 appear to be handled, which is good.

Next, let's dig a little deeper to see what happens on errors

</details>

<details>
<summary>When softdevice disabled, and writing to flash fails</summary>

<P>

When the softdevice is disabled, no event is generated and the call to `sd_flash_write()` occurs synchronously.  Therefore, only the code in the function needs to be reviewed.
https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/8dbe7d950363dc17ed6c571be8dd66f53507feca/cores/nRF5/nordic/softdevice/s140_nrf52_6.1.1_API/include/nrf_soc.h#L923-L924

#### BUG 1 -- Code does not report write failures
Here, if the call to `sd_flash_write()` fails with any status other than NRF_ERROR_BUSY, then the only thing that occurs is that a debug log message is generated via VERIFY_STATUS().  Critically, the parameter `len` is still returned to the caller, and the return value of the function is defined to represent the number of bytes written.
https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/8dbe7d950363dc17ed6c571be8dd66f53507feca/libraries/InternalFileSytem/src/flash/flash_nrf5x.c#L127-L131
</details>

<details>
<summary>When softdevice disabled, and writing to flash fails</summary>

<P>

When the softdevice is enabled, the call to `sd_flash_write()` is asynchronous.  Therefore, if the call to `sd_flash_write()` returns NRF_SUCCESS, that only means the command was queued, not that it actually succeeded.  The code must then wait for the event to be generated to know if the write succeeded or failed.  This means that there are two failure paths to consider.  First, when the call to `sd_flash_write()` fails, and second, when that succeeds, but the event generated is `NRF_EVT_FLASH_OPERATION_ERROR`.  **Both appear to have bugs.**

**BUG2 -- Code does not report failures from `sd_flash_write()`**
Identical to bug 1, but with softdevice enabled.  It's also unclear from the documentation if an event will be generated when the call to `sd_flash_write()` fails with an error.  If not, then this could also cause a hard-lock when this function attempts to take the semaphore.

**BUG3 -- Code does not detect or report failures from  event**
If the call to `sd_flash_write()` returned NRF_SUCCESS, then the command was queued, and the success or failure is determined by the event generated (`NRF_EVT_FLASH_OPERATION_SUCCESS` or `NRF_EVT_FLASH_OPERATION_ERROR`).

The overall event handler is `adafruit_soc_task()`:
https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/8dbe7d950363dc17ed6c571be8dd66f53507feca/libraries/Bluefruit52Lib/src/bluefruit.cpp#L686-L712

`adafruit_soc_task()` calls `flash_nrf5x_event_cb()` at line 711.  That function is:
https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/8dbe7d950363dc17ed6c571be8dd66f53507feca/libraries/InternalFileSytem/src/flash/flash_nrf5x.c#L37-L41

Critically, neither of those functions in any way stores an indication that the write failed.  This prevents `fal_program()` from propagating the failure, and is thus appears to be a bug.
</details>

<details>
<summary>Possible semaphore race condition?</summary>

<P>

OK, this is not as sure, as there may be other FreeRTOS restrictions that prevent this.  However, it at least smells bad....
1. `fal_program()` only takes the semaphore **AFTER** the write command has been queued.
2. the soc event handler unconditionally calls xSemphoreGive() for either `NRF_EVT_FLASH_OPERATION_SUCCESS` or `NRF_EVT_FLASH_OPERATION_ERROR` events.
3. a semaphore cannot be given unless it was first taken.  From the [FreeRTOS documentation](https://www.freertos.org/a00123.html):
> An error can occur if there is no space on the queue to post a message – indicating that the
> semaphore was not first obtained correctly.

Therefore, the soc event handler presumes that the semaphore is already owned.  However, the semaphore is not taken until **after** the command is queued.  Therefore, it appears that the following might be possible to occur:
1. write is queued from `fal_program()`
2. **_callback for soc event interrupts ..._**
3. callback `flash_nrf5x_event_cb()` calls `xSemaphoreGive()`, which returns an error because there is no owner, but returns silently.
4. **_callback for soc event completes ..._**
5. `fal_program()` acquires semaphore
6. deadlock?

</details>

In short, it appears that InternalFileSystem currently **_completely ignores flash errors_**.

More than this, the file system is not written in a way that would be robust against power failures (or other failures to write a page), because it erases the existing page's data and writes the new page data in the same sector.  This is an inherently problematic design, if you want a file system that is robust against power failure events.

Let me see if I can generate a branch with a  patch file that can at least catch whether these errors are occurring, or if there are additional problems somewhere else.....

henrygab, 2019-10-05:
@hathach -- I've written a fairly scathing analysis in the above few posts.  That analysis suggests that the InternalFileSystem library is architecturally not reliable to power failures, and that the currently implementation doesn't handle any write errors … at all.  I'd appreciate your review of the above analysis, as I'm hoping I missed something and am entirely wrong in my analysis....

shaoyuan1943, 2019-10-07:
Hi @henrygab 
Thanks for ur nice job!

henrygab, 2019-10-09:
I've opened a new bug (#350), based on the above analysis, with a title that more accurately reflects the cause.  While ARMmbed's LittleFS is robust to power failures and failed writes, it must be told when a write fails.  Hopefully, having been provided this deeper analysis, @hathach will have the information needed to rework the InternalFS library to propagate errors.

henrygab, 2019-10-09:
@shaoyuan1943 -- Do you have a reliably repro for this bug?

The issue is more subtle than I originally thought.  But, I have additional possible thoughts, based on your log above, related to multiple tasks calling into LFS simultaneously.  It's not clear to me if the InternalFileSystem code can be called safely in parallel. (See [re-entrant](https://en.wikipedia.org/wiki/Reentrancy_%28computing%29) on Wikipedia.)

If you have the ability to reliably repro this bug, please let me know.  Then, I may be able to create a private branch that adds a mutex at the InternalFileSystem callbacks, to ensure that even if LFS attempts to write in parallel, that only one executes at a time.

shaoyuan1943, 2019-10-10:
@henrygab I am not sure if I can reproduce this bug, but I will continue to try it. I plan to spend this weekend trying to reproduce this issue :)

henrygab, 2019-10-10:
@shaoyuan1943 -- Some questions to try to repro this myself:
1. Can you provide a complete sketch?  (what you provided does not compile for me)
2. You report using BSP 0.11.1 ... is there a reason you are not updated to 0.14.0?
3. The partial sketch you provided appears to suggest you are running on battery.  Is that right?
4. If you can reliably repro, and are running on battery, does it repro when not on battery?

These will help in attempts to create a repro of the issue.

( FYI, running on battery is OK .. LittleFS should still not end up with corruption .. but knowing it's on battery could help create a repro, such as by lowering the power supply voltage to "emulate" partially battery, or configuring power supply to have low max amperage, to force flash errors to be more common. )

BTW, a [prototype](https://github.com/henrygab/Adafruit_nRF52_Arduino/compare/SerializeFlashPrototype) that at least compiles now exists, where the change is to force serialization of all access to the flash API.  Also, this prototype should* log to serial if this situation ever occurs.
_(*) untested_

See the main concept in the [single file commit](https://github.com/henrygab/Adafruit_nRF52_Arduino/commit/93f61f25bcd21b3199828884108de99a0f04fcac):
`AdaFruit/hardware/nrf52/0.14.0/libraries/InternalFileSytem/src/flash/flash_cache.c`

shaoyuan1943, 2019-10-10:
Hi @henrygab 
1. I will provide complete sketch later.
2. The BSP version is 0.11.1, I will test it on 0.14.0 later.
3. 4. I'm not sure whether to use USB or Li battery.

henrygab, 2019-10-11:
@shaoyuan1943 -- a stable power connection repro would be best.  If it only repro's when running on battery, that's OK, but knowing that it does not repro on stable power will be helpful.

henrygab, 2019-10-14:
@shaoyuan1943 -- Have you been able to reproduce the problem?  If so, have you re-compiled and tried to repro with the following single-file change (also at debug level 2)?

[AdaFruit/hardware/nrf52/0.14.0/libraries/InternalFileSytem/src/flash/flash_cache.c](https://github.com/henrygab/Adafruit_nRF52_Arduino/commit/93f61f25bcd21b3199828884108de99a0f04fcac)

shaoyuan1943, 2019-10-15:
@henrygab I did not reproduce this problem, I will try your modified:)

shaoyuan1943, 2020-01-01:
Hi~ @hathach @henrygab  can I close this issues now?

hathach, 2020-01-01:
It is up to you, you can close it only if you couldn’t reproduce the issue

closed 2020-01-06 by hathach

esev, 2025-01-19:
Drive-by comment many years later. 

```
11:07:35.692 -> [SOC ] NRF_EVT_FLASH_OPERATION_ERROR
11:07:36.308 -> [BLE ] BLE_GAP_EVT_DISCONNECTED : Conn Handle = 1
11:07:36.308 -> [GAP ] Disconnect Reason 0x08
11:07:36.308 -> [SOC ] NRF_EVT_FLASH_OPERATION_SUCCESS
11:07:36.475 -> [SOC ] NRF_EVT_FLASH_OPERATION_SUCCESS
11:07:36.509 -> assertion "head >= 2 && head <= lfs->cfg->block_count" failed: file "C:\Users\cc\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.11.1\libraries\Adafruit_LittleFS\src\littlefs\lfs.c", line 1144, function: lfs_ctz_find
```

These things seem to be related. Specifically a BLE timeout disconnect during a flash operation can cause the flash operation to fail.

This causes corruption in LittleFS due to the `block_size` being configured smaller than the flash page size.
https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/4dcfa3b7960a6dafec6736a5dbea255b0946be83/libraries/InternalFileSytem/src/InternalFileSystem.cpp#L111

LittleFS is only able to recover & relocate 128 bytes from the failing 4096/2048 physical flash page.

https://github.com/adafruit/Adafruit_nRF52_Arduino/pull/838 will be able to help in this situation. But due to the above, I think LittleFS won't be able to recover from a non-retryable flash error.

PR #397 (2020-01-06): Serialize lfs
Fix #350
Fix #325
Fix #227
Fix #222
Fix #393

@hathach -- This PR supercedes #372 (Guard against race conditions in Flash FS cache), as #72 was not really providing a proper fix of the underlying problem:

LFS itself is [not safe](https://github.com/ARMmbed/littlefs/issues/352) to call simultaneously from multiple contexts.

This PR solves this by wrapping the InternalFS API with a non-recursive mutex, as recommended.

I realize the PR uses newly declared wrapper functions.  I tried to avoid this, but the changes got messy quite quickly, mostly due to some of the macros used to v [... cut]