adafruit/Adafruit_nRF52_Arduino #334: Stacks  0.11.1  to 0.13.0  do not work on feather nrf52832.

jps2000, 2019-08-28:
**Describe the bug**
stacks later than 0.11.0 (to 0.13.0) do not work on feather nrf52832.
Tested with bleuart
https://forums.adafruit.com/viewtopic.php?f=57&t=155536
(does not get any response there)
After  booting the blue LED does not start blinking / advertising.
Version 0.11.0 is latest that works

**Set up (please complete the following information)**
feather nrf52832 
software bluefruit52 0.11.1  to 0.13.0
Arduino IDE 1.8.5
debug Level 0.
When selected higher levels continuous reboot occurs. red led "flashing" DFU pin goes periodically low

**To Reproduce**
upload bleuart example

**Expected behavior**
blue led should blink , should advertise  and device connectable in bluefruit LE app

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

coral, 2019-08-30:
Same issue.

jps2000, 2019-08-31:
I tried now Arduinio IDE 1.8.9 and now it works
Not clear why and not mentioned.

closed 2019-08-31 by jps2000