adafruit/Adafruit_nRF52_Arduino #335: [Feature] Additional Device Information

elral, 2019-08-29:
For my project I need to set additional device parameters. To do this I added some methods to BLEDis.cpp and BLEDis.h. It would be great if these additions could be added to the masterbranch.  Unfortenutely I do not know how to submit a pull request, therefore I addded my code here.

Thanks and regards

[BLEDis.zip](https://github.com/adafruit/Adafruit_nRF52_Arduino/files/3554159/BLEDis.zip)

Ralf

hathach, 2019-08-29:
submit a PR is very simple, it is much easier for you and us than attaching zip file. Just create a folk, modify the code then submit PR. Here is the tutorial from github.
https://help.github.com/en/articles/creating-a-pull-request

elral, 2019-08-31:
Thanks, I forked it and changed the files, but "unfortenutely" more than for this Pull Request. It seems that no files can be choosen for a Pull Request, so I have to start again. Will start next week.

hathach, 2019-08-31:
You dont need to choose file, only need to modify, add file to github then commit and push. Then create the pr, github will do the rest for you

commit 343c9467 (2019-09-26): Merge pull request #341 from adafruit/develop

follow up to pr #336 , close #335
closed 2019-09-26 by hathach